ev3dev/ev3dev #551: Execute files from brickman

0schi, 2016-02-15:
Hello,

I'm new to ev3/ev3dev but I managed to write python-files and execute them from terminal via ssh/usb.
But now I need to execute those files from brickman itself because I need the ev3 to move freely.

First I just wanted to execute them by clicking the file.py in the file browser. Which led to a no-executable-file-errormessage.
Then I tried to make it executable by shebanging. So i included `#!/usr/bin/env python` as first statement in my python file...
If I try to execute this file, I'll get a message there wouldn't be such a file or directory.

I don't know what to try next :/

I hope for help. My masterthesis is at stake ^^'

Some words to the file I try to execute:
It's an testfile to get some information from a Wii Balance Board via blootooth. It imports a wiiboard.py from Nedim Jackman and some ev3dev specific packages to move the ev3.
Basically it creates a new Wiiboard-instance, connects the balance board to the ev3 via bluetooth and gets and display some parameters from the device.

ddemidov, 2016-02-15:
See #512

0schi, 2016-02-15:
Thanks for pointing those issues out.
BUT I allready tried what was written in those issues:
- I set `#!/usr/bin/env python` on the beginning
- I wrapped the top level code with `if __name__ == __main__`
- I made it executable using `chmod u+x examply.py`

I still get a message it that there ist no such file or directory.... 

I'm not sure but is it relevant where I put the wiiboard.py which wasn't preinstalled?

ddemidov, 2016-02-15:
Does it say what file exactly it can not find? If you create a simple hello-world python file, will it work for you? Are you using the default `robot` user and placing the files into `/home/robot`?

If `wiiboard.py` is in the same directory as the main file, you should be able to import it.

0schi, 2016-02-15:
It doesn't say what it can't find.
It's just saying `:No such file or directory` - hard to tell since it isn't visible for more than a second.

I just tested the helloworld file. It's the same result/message.

I'm using the default user and place all files in the same place `/home/robot`

ddemidov, 2016-02-15:
How do you launch the files from command line? If its `python file.py`, can you see if you can reproduce the error by running `./file.py`? Also your shebang line seems correct, but I would also try with `#!/usr/bin/python`.

0schi, 2016-02-15:
From comman line i used `python example.py` to launch the programm. Worked all the time. 

I tried `#!/usr/bin/python` in this helloworld:
```python
#!/usr/bin/python
import time
print "Hello World!"
time.sleep(2)
```
Here were two errors: It couldn't find the import command for time at line 2 and couldn't find the print command at line 3.
BUT it the text was visible for arround 2 seconds (time.sleep(2)?)

ddemidov, 2016-02-15:
The problem may be with line ending symbols (which could happen if you used created the files with a windows editor and moved them to ev3):
```
$ cat hello.py 
#!/usr/bin/env python
print('Hello World')
$ ./hello.py 
: No such file or directory
$ dos2unix hello.py 
dos2unix: converting file hello.py to Unix format...
$ ./hello.py 
Hello World
 ```

0schi, 2016-02-15:
Well I used Notepad++ on Windows to create the files and transfered them via WinSCP to the EV3.

ddemidov, 2016-02-15:
Does converting from dos to unix format help?

0schi, 2016-02-15:
I needed to install dos2unix first. But now its functioning properly :)
Thank you very much for your help! You just made my masterthesis a success (hopefully)

closed 2016-02-15 by 0schi

BertLindeman, 2016-02-15:
@0schi Notepad++ has a button `Show all characters` that shows the lineendings, so `LF` or `CRLF`
and Notepad++ has a conversion tool in the menu: 
`Edit` -> `EOL Conversion` -> `Convert to UNIX format`
or Windows or Mac format.

zhuqiaorong, 2017-05-18:
Yes,it works!Awesome!
#!/usr/bin/env python3
  
and then
a.  apt-get install dos2unix
b.  dos2unix test.py
c.  chmod +x test.py

myrlan001, 2017-11-09:
Thank you. I had same problem.
Installing dos2unix and converting the file solved the problem.
If I upload the file from ubuntu to ev3dev will I also have to convert the file?

feiniao1933, 2018-07-21:
yes! It works!
adding
#!/usr/bin/env python3
at the beginning of the py file
be careful not to miss the "3" after "python"
maybe for some certain ev3dev version

yvcode, 2018-07-23:
Amm... It didnt work to me... Help?