ev3dev/ev3dev #552: seeking and reading .../direct attribute doesn't refresh the data

jdchristensen, 2016-02-16:
Here's a simple python program that keeps seeking and reading a byte from a given file:

    import sys
    fd = open(sys.argv[1], 'rb')
    for i in range(10):
        fd.seek(66)
        print ord(fd.read(1))

With the Absolute IMU gyro sensor, this should produce some of the tilt data.  And yet, if I run it like this:

`python seektest.py /sys/devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0011/lego-sensor/sensor1/direct`

I get the same value 10 times, even with the device moving, and even if I put a 0.1s delay in the loop.

If I close and reopen the file each time through the loop, the values change, but this is very slow.

dlech, 2016-02-16:
You have made the common mistake of not disabling buffering. Try this.

```python
import sys
fd = open(sys.argv[1], 'rb', 0)
for i in range(10):
    fd.seek(66)
    print ord(fd.read(1))
```

closed 2016-02-16 by dlech

jdchristensen, 2016-02-16:
Thanks, it works fine with that change!