ev3dev/ev3dev #553: Add an issue template

WasabiFan, 2016-02-17:
GitHub now supports [issue templates](https://github.com/blog/2111-issue-and-pull-request-templates). We should add one to this repo so that there's a more obvious indication of what every report should have.

dlech, 2016-02-25:
We'll do this after the next image release that will include the `ev3dev-sysinfo` script.

dlech, 2016-03-07:
Maybe we should do this a bit sooner. Does anyone have a good bug reporting template.

Here is what debian uses:

   * What led up to the situation?
   * What exactly did you do (or not do) that was effective (or
     ineffective)?
   * What was the outcome of this action?
   * What outcome did you expect instead?

Mozilla asks:

* What did you do? (steps to reproduce)
* What happened? (actual results)
* What should have happened? (expected results)

Some ev3dev-specific things we need to know are what image file was used. what platform, what kernel version. If the problem has to do with something other than the kernel, then also a name and version of the package.

dlech, 2016-03-07:
Here is a start...

Tips:

* Use the toolbar buttons above to insert quotes, code, lilnks, lists, etc. 
* When pasting more than one line of code or the output of a terminal window, use a "codefence" like this:

        '''
        <paste stuff here>
        '''

### System information:

[Delete everything that does not apply]

* I am using [LEGO MINDSTORMS EV3, Dexter Industries BrickPi, Mindsensors.com PiStorms, Fatcatlab EVB] with [Rapsberry Pi Model [A, A+, B, B+, 2B], BeagleBone [Black, White, Green]].

* I installed ev3dev using the [ev3dev-???.img] file.

* My kernel version is <paste the output of `uname -r` here>.

### About my issue:

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

WasabiFan, 2016-03-07:
- If you write the tips section as an XML comment, it won't show up in the rendered view -- I think that would make it a bit cleaner.
- It isn't immediately obvious what the [brackets] mean. I think there should be a longer explanation of why they are seeing this content and what we expect them to do.
- Typo in first bullet: lilnks
- We should add an explanation of getting the kernel version for now until we an image with the sysinfo tool is released
 - Explain that this only applies for bugs or problems -- feature suggestions and such can ignore this content.

dlech, 2016-03-20:
Thanks for the feedback. This is live now.

https://raw.githubusercontent.com/ev3dev/ev3dev/master/ISSUE_TEMPLATE.md

https://github.com/ev3dev/ev3dev/blob/master/ISSUE_TEMPLATE.md

closed 2016-03-20 by dlech