ev3dev/ev3dev #554: PiStorms how to show a video in the screen and using HDMI?

jabrena, 2016-02-21:
Hi @dlech,

I would like to know how to play a video on PiStorms using the Screen and Using the HDMI output from the raspberry Pi. Is it possible?

Cheers

Juan Antonio

dlech, 2016-02-21:
I was able to get a video to play on PiStorms using this tutorial: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/playing-videos

jabrena, 2016-02-21:
Great!!!

[image]

how to watch the video using the HDMI output?

Cheers

dlech, 2016-02-21:
Use `FRAMEBUFFER=0`? I've never actually done this, so you will have to figure it out for yourself. Surely there is some web page about this already.

jabrena, 2016-02-21:
Hi, I tried but I receive the following error:

``` bash
robot@ev3dev:~$ FRAMEBUFFER=0
robot@ev3dev:~$ sudo mplayer -framedrop bigbuckbunny320p.mp4
MPlayer2 2.0-728-g2c378c7-4+b1 (C) 2000-2012 MPlayer Team
Cannot open file '/root/.mplayer/input.conf': No such file or directory
Failed to open /root/.mplayer/input.conf.
Cannot open file '/etc/mplayer/input.conf': No such file or directory
Failed to open /etc/mplayer/input.conf.

Playing bigbuckbunny320p.mp4.
Detected file format: QuickTime / MOV (libavformat)
[lavf] stream 0: video (h264), -vid 0
[lavf] stream 1: audio (aac), -aid 0, -alang und
Clip info:
 major_brand: mp42
 minor_version: 0
 compatible_brands: mp42isomavc1
 creation_time: 2013-11-26 16:05:17
 encoder: HandBrake 0.9.9 2013052900
Load subtitles in .
vo: couldn't open the X11 display ()!
vo: couldn't open the X11 display ()!
vo: couldn't open the X11 display ()!
[gl] OpenGL 3.x context creation not implemented.
vo: couldn't open the X11 display ()!
SDL SetVideoMode failed: OpenGL not available
vo: couldn't open the X11 display ()!
[VO_SDL] Using driver: fbcon.
vo: couldn't open the X11 display ()!
```

Next week, I will try again this idea.

Cheers

dlech, 2016-02-21:
You for got `export`.

    export FRAMEBUFFER=0

Also, you should not need `sudo`. Just make sure you are on a terminal running on the HDMI already.

jabrena, 2016-02-22:
nope, with this configuration the video is showed on the terminal.

[image]

``` bash
robot@ev3dev:~$ export FRAMEBUFFER=0
robot@ev3dev:~$ mplayer -framedrop bigbuckbunny320p.mp4
```

dlech, 2016-02-22:
This is expected if you run the command on a remote (ssh) terminal. You have to be logged on to a virtual console on the the HDMI output itself.

jabrena, 2016-03-04:
Is it possible to test the same example on the EV3 Brick?

I tried with:

``` 
sudo SDL_VIDEODRIVER=fbcon SDL_FBDEV=/dev/fb0 mplayer -vo sdl -framedrop bigbuckbunny320p.mp4
```

But I couldn't do it.

dlech, 2016-03-04:
Afik, SDL does not support monochrome (1-bit per pixel). You have to use something else. Maybe you could ask this person how they [played video on the EV3](https://www.youtube.com/watch?v=f8I2AMFNUoY).

rookyew16098, 2017-06-25:
Maybe AAlib with a 1x1 font?

closed 2017-07-25 by dlech