ev3dev/ev3dev #555: Incorrect HiTechnic-Angle-Sensor readout

harmooni, 2016-02-25:
Hello,

I'm facing problems using the HiTechnic Angle Sensor. Only the 'ANGLE' mode returns correct values.
The output of the remaining modes can't be right. Maybe it's an endianness problem.

Output of the sensor:

```
ANGLE:  41
ANGLE-ACC:  -2013265921
SPEED:  -31489 

ANGLE:  38
ANGLE-ACC:  -2097152001
SPEED:  -31489 

ANGLE:  37
ANGLE-ACC:  2147483647
SPEED:  -32770 

ANGLE:  37
ANGLE-ACC:  2130706431
SPEED:  -33026 

ANGLE:  36
ANGLE-ACC:  2130706431
SPEED:  -33026 

ANGLE:  38
ANGLE-ACC:  -2063597569
SPEED:  -32257 

ANGLE:  42
ANGLE-ACC:  -1946157057
SPEED:  -29441 
```

Attached is the documentation of the sensor.

I tested it on ev3-ev3dev-jessie-2015-12-30.img - kernel version: 3.16.7-ckt21-9-ev3dev-ev3
and on ev3-ev3dev-jessie-2015-09-13.img

Many thanks in advance.

[image]

dlech, 2016-02-25:
Looks like you are right about the endianness. Relevant code is [here](https://github.com/ev3dev/lego-linux-drivers/blob/c42f807ce2cf62f960c5c814a11298ec6e67c01d/sensors/nxt_i2c_sensor_defs.c#L1005). 

For the `SPEED` mode, we have `LEGO_SENSOR_DATA_S16_BE` which should fix that mode. However, for the `ANGLE-ACC` mode, we don't have `LEGO_SENSOR_DATA_S32_BE`, so it will have to be created.

Would you be up to the task?

harmooni, 2016-02-26:
Wow, that was fast. 
Yes, I can try doing that. If I have questions, I hope I can ask you.

harmooni, 2016-02-27:
Hello,

A short update:

- I added support for LEGO_SENSOR_DATA_S32_BE and the 'ANGLE-ACC' mode is working now.

- I realized that the 'ANGLE' mode was'nt correct after all: Only the unscaled 2° increments were shown. I added an extra scale function for this mode.

- I still have problems with the 'SPEED' mode (using LEGO_SENSOR_DATA_S16_BE). Spinning the axle counterclockwise the output looks fine. If it's rotating clockwise there is still an overflow.

Axle rotates counterclockwise:
```
ANGLE 178
ANGLE-ACC -101
SPEED 0 

ANGLE 3
ANGLE-ACC -101
SPEED -191 

ANGLE 3
ANGLE-ACC -300
SPEED -9 

ANGLE 338
ANGLE-ACC -300
SPEED 0 
```

Axle rotates clockwise:
```
ANGLE 131
ANGLE-ACC -469
SPEED 0 

ANGLE 285
ANGLE-ACC -351
SPEED -65346 

ANGLE 298
ANGLE-ACC -340
SPEED -65346 

ANGLE 299
ANGLE-ACC -339
SPEED 0 
```

dlech, 2016-02-27:
It looks like we have the wrong starting address for the SPEED mode [here](https://github.com/ev3dev/lego-linux-drivers/blob/c42f807ce2cf62f960c5c814a11298ec6e67c01d/sensors/nxt_i2c_sensor_defs.c#L1032). In the code it is 0x46, but according to the document you posted, it should be 0x48.

harmooni, 2016-02-27:
Yes, you're right. I've already changed the address to  0x48. Prior to this the speed was not equal to zero even though the axle wasn't moving.

dlech, 2016-02-27:
So still not working after you have make the change? Can you post your code somewhere or start a pull request?

dlech, 2016-02-27:
I don't think `LEGO_SENSOR_DATA_S16_BE` has actually ever been tested. Try changing the code [here](https://github.com/ev3dev/lego-linux-drivers/blob/6941249ebac882a8f87544bebff232cdd2c293d8/sensors/lego_sensor_class.c#L373) from `ntohs` to `be16_to_cpu`. Likewise, use `be32_to_cpu` for the 32 bit function.

harmooni, 2016-02-27:
Yes, the source of the problem has to be there. When I converted the sensor data manually (SPEED2), the output seems correct (see below). Why it' s working for 32-bit, I can't say. 
I'll modify the code as you suggested.
```
ANGLE 155
SPEED -65488
SPEED2 316 

ANGLE 155
SPEED -65313
SPEED2 224 

ANGLE 196
SPEED 0
SPEED2 224 

ANGLE 193
SPEED -3
SPEED2 224 

ANGLE 96
SPEED -99
SPEED2 -99 

ANGLE 98
SPEED -328
SPEED2 -328 
```

harmooni, 2016-02-27:
I replaced ntohs with be16_to_cpu, but unfurtunately this didn't solve the problem. The output is still the same.

dlech, 2016-02-27:
That is strange. Since we are dealing with pointers, I wonder if we could use `be16_to_cpup` instead. 

http://www.bruceblinn.com/linuxinfo/ByteOrder.html

How did you manually convert the data?

harmooni, 2016-02-28:
You can see [here](https://gist.github.com/harmooni/935f6dae32f1950451d1) how I converted the data manually.

I've found the bug by now. The overflow happens  [here](https://github.com/ev3dev/lego-linux-drivers/blob/6941249ebac882a8f87544bebff232cdd2c293d8/sensors/lego_sensor_class.c#L392) while converting from raw data range to scaled data range.
As the computations are made step by step, the overflow happens at 
```
*value = (*value - mode_info->raw_min) * ((mode_info->si_max - mode_info->si_min)
```
This seems to be computed differently in the 32-bit case. You can find [here](https://gist.github.com/harmooni/1549653fc3ebd8cc3716) my simple test program to reproduce the overflow. Putting 
```
(mode_info->si_max - mode_info->si_min) / (mode_info->raw_max - mode_info->raw_min)
```
in brackets solves the problem. 

Another question: I don't understand the if statement. Shouldn't it look more like this:
```
if (mode_info->raw_min != mode_info->si_min || mode_info->raw_max != mode_info->si_max)
```

rhempel, 2016-02-28:
Scaling issues are very easy to run into on 32 bit platforms - nice catch!

dlech, 2016-02-28:
Really what the if statement is checking for (in a very lazy way) is that `raw_min == 0 && raw_max == 0`. In other words, the values have not been set at all. I like your suggestion better, so feel free to change it.

closed 2016-03-04 by harmooni

cho934, 2016-03-25:
If I understand well, this bug is always in ev3-ev3dev-jessie-2015-12-30.img - kernel version: 3.16.7-ckt21-9-ev3dev-ev3, and will be solved in the next release of the kernel ?

Is it possible to test it before ?

dlech, 2016-03-25:
Yes, this is what the `fix committed` tag means. You can compile a kernel yourself if you want to test, but be aware there have been significant changes to motors, so motors won't work with any existing library.

cho934, 2016-03-25:
Thank you. I would like to use external encoders to avoid slippage of my robot.
So it means that it will be difficult to use it during the French cup of robotic scheduled begining of May...

dlech, 2016-03-25:
Yes, it will probably be some time before we do an official kernel release. You could build a kernel yourself from the git tag of the last release and just cherry-pick the relevant patches if you don't want the motor changes yet.

dlech, 2016-06-15:
I think this was actually fixed in the `11-ev3dev` release.
We have just released [12-ev3dev](http://www.ev3dev.org/news/2016/06/14/Kernel-Release-Cycle-12/) kernel.