ev3dev/ev3dev #556: Our projects page grid is horribly misaligned

WasabiFan, 2016-02-27:
Due to some shortcuts that were taken when creating the projects page (not naming names :wink:), the variable length of project descriptions has caused the grid to become very badly aligned.

Involved page: http://www.ev3dev.org/projects/

WasabiFan, 2016-02-27:
Also, I think we should consider reenabling issues on the site repo to track site-specific formatting issues. We would need to make it clear that it is only for styling and formatting, though.

dlech, 2016-02-27:
I like having all of the issues in one place. But for internal stuff, it does make sense to maybe have a separate issue tracker for the website to reduce the noise here. I'll go ahead and turn it on.

WasabiFan, 2016-02-27:
Should I reopen the existing site-related issues over there now?

dlech, 2016-02-27:
Whatever you like.

WasabiFan, 2016-02-27:
Migrated to ev3dev/ev3dev.github.io#110

closed 2016-02-27 by WasabiFan