ev3dev/ev3dev #557: Comprehensive compatibility/comparison list for EV3, BrickPI and PiStorms?

dragoncube, 2016-02-27:
Is there any list of supported commands/attributes for mentioned devices? I'm trying to understand which platform at this point is better to use BrickPI or PiStorms and such list would be very useful to have.

dlech, 2016-02-27:
Right now, the best we have is [this](https://github.com/ev3dev/ev3dev/issues/412#issuecomment-154242776).

dlech, 2016-02-27:
Also, compare open issues for [PiStorms](https://github.com/ev3dev/ev3dev/issues?q=is%3Aopen+is%3Aissue+label%3APiStorms) and [BrickPi](https://github.com/ev3dev/ev3dev/issues?q=is%3Aopen+is%3Aissue+label%3ABrickPi).

dragoncube, 2016-02-29:
btw, BrickPi does not support stop_commands "break" and "hold" which is mentioned here: http://www.ev3dev.org/news/2015/05/13/raspberry-pi-image-release/ . Is this limitation due to the issues discussed here: http://www.dexterindustries.com/topic/accuracy-of-rotating-a-motor/ ? Is there any chances to address it without firmware changes?

dlech, 2016-03-01:
"brake" (passive breaking) is not possible without a firmware change. "hold" could be implemented without firmware changes. I'm not sure how good it would be due to latency issues in getting position feedback, but it would probably be decent.

There are also issues with EV3/UART sensors on the BrickPi that require firmware changes to fix.

dragoncube, 2016-03-01:
As I understand "hold" is much more valuable, since without it you can't use effectively use such commands as "run-to-abs-pos" and "run-to-rel-pos"

bmegli, 2016-10-12:
See also #640 

I know little about PiStorms.

For BrickPi:

Those I2C sensors (of many more) will not work currently on BrickPi+:

@kortschak wrote:
>I count 7: "ht-nxt-smux", "mi-xg1300l", "ms-absolute-imu", "ms-ev3-smux", "ms-nxtmmx", "ms-8ch-servo", "ms-pps58-nx".

But it's possible that they will work in the far future.

dlech, 2016-10-17:
Work in progress: http://127.0.0.1:4000/docs/platform-comparison/

Still needs more info on motor and sensor quirks between platforms.

closed 2017-07-25 by dlech