ev3dev/ev3dev #558: Can't change speed_sp anymore why motor is running

ensonic, 2016-02-29:
Is this intentional? 
```bash
cd /sys/class/lego-port/port5/outB:lego-ev3-l-motor/tacho-motor/motor0
echo "100" >speed_sp
echo "run-forever" > command
# about 100
cat speed
101
# no effect
echo "200" >speed_sp
# now it changes
echo "run-forever" > command
echo "stop" > command
```

From e.g. the python api, this is a bit awkward. To start a motor at some speed I can do:
```python
m.run_forever(speed_regulation_enabled='on', speed_sp=200)
```

but when I want to change speed, I can't just do:
```python
m.speed_sp = 300
```
but need to do something like:
``` python
if m.state:
  m.run_forever(speed_regulation_enabled='on', speed_sp=300)
else:
  m.speed_sp = 300
```

WasabiFan, 2016-02-29:
I believe that this is intentional -- historically @dlech has recommended the `run_direct` command instead if you want to change the value dynamically.

dlech, 2016-02-29:
Yes, this intentional. You set all of the parameters (setpoints), then send a command to make the parameters take effect.

What is the use case for changing the speed when the motor is not running?

dlech, 2016-02-29:
By the way, if you haven't been following along, @rhempel is working on a major motor driver overhaul in ev3dev/lego-linux-drivers#23. This will certainly affect OpenRoberta, so if you need the motor drivers to work a certain way, speak up now.

ensonic, 2016-02-29:
I can handle this from OpenRoberta. I'll send a PR for the pythondocs to make this more obvious.

closed 2016-02-29 by ensonic

ensonic, 2016-03-01:
Okay just to be sure:

For unregullated motor controls: I should always be using ```run_direct```, right? This makes the api asymetric though as for ```run_direct``` the [docs](http://python-ev3dev.readthedocs.org/en/latest/motors.html#tacho-motor) claim, that "Unlike other run commands, changing duty_cycle_sp while running will take effect immediately."

``` python
    def turnOnRegulatedMotor(self, port, value):
        value *= 10.0
        self.cfg['actors'][port].run_forever(speed_regulation_enabled='on', speed_sp=int(value))

    def turnOnUnregulatedMotor(self, port, value):
        value *= 10.0
        self.cfg['actors'][port].run_direct(duty_cycle_sp=int(value))

    def setRegulatedMotorSpeed(self, port, value):
        value *= 10.0
        m = self.cfg['actors'][port]
        if m.state:
          m.run_forever(speed_regulation_enabled='on', speed_sp=int(value))
        else:
          m.speed_sp = 300

    def setUnregulatedMotorSpeed(self, port, value):
        value *= 10.0
        self.cfg['actors'][port].duty_cycle_sp = int(value)
```

rhempel, 2016-03-01:
First, let's break the command set into `run-direct` and everything else.

`run-direct` gives you an unregulated motor, only the `duty_cycle_sp` can be controlled, and it can be changed at any time. You still get the position feedback, but no ramping.

All other modes give you a speed regulated motor, with ramping, run to position, and tacho feedback.  You can change any setpoint at any time, but it will not take effect until the next motor command is sent.

I'll review the docstring changes this evening and merge then.

dlech, 2016-03-01:
@ensonic, yes your code looks correct.