ev3dev/ev3dev #559: Ideas for a Linux Course

jabrena, 2016-03-01:
Good morning @dlech, 

This month, I will teach a basic course about Linux with EV3Dev.

The main points to explain are:

- Installation of EV3Dev
- Remote connection: SSH
- Linux Filesystem
- Processes
- Users & Groups
- Network
- Utilities
- Package Manager
- EV3Dev: Interaction with /sys/class

Any new comment to add?

Cheers

Juan Antonio

rhempel, 2016-03-01:
Pull out processes, users and groups, utilities and package manager. Add a brief section on Filezilla.

Get the users from install, to ssh/FileZIlla, maybe network, and then right to interaction with sys/class to avoid boring them with stuff they don't need right away to program a robot.

Give them access to the brick, and have them blinking lights within 15 to 20 minutes - then move on to touch sensor to control a light, then motors.

jabrena, 2016-03-01:
Hi @rhempel, the activity with Filezilla could be very practical to test some Scripts developed in several languages for example.

Teenagers in general like quick feedback so some activities to use motors, leds could be great. Later, I could add some activities with some sensors.

But I think that top and kill could be useful to learn, that commands are very practical. Package manager is another interesting feature from Debian but I recognize that it should be added at the end or avoid this topic in a first iteration.

More comments?

Cheers

Juan Antonio

dlech, 2016-03-01:
If the class is more focused on learning Linux than learning robots, then yes, you should spend some time on basic commands like `ls`, `cat`, `grep`, `ps`, `top` and pipes and redirection and that sort of thing. Another good thing to do would be to write a script and make it executable. Environment variables are another important fundamental topic.

jabrena, 2016-03-04:
A beginning...
http://ev3dev-lang-java.github.io/linux-course/#/

:)

closed 2017-07-25 by dlech