ev3dev/ev3dev #560: my ev3 wont boot into ev3dev

spike11302000, 2016-03-06:
every time i try to boot into the sd it will not boot

jabrena, 2016-03-06:
Good morning @spike11302000,
 
maybe the sd card had some problem. Did you test to change your sd card? Are you checked that the sd card was burned with the image?

Juan Antonio

harmooni, 2016-03-06:
Hello @spike11302000,

(Just like @jabrena suggested) I also recommend plugging the SD Card in your PC, and check if you can see the ev3dev system.
Which operating system did you use to flash your SD Card?

dlech, 2016-03-06:
Did you unzip the image file before writing to the SD card?

spike11302000, 2016-03-07:
yes i have unzip and try different sd cards

but i can boot into jelos. [http://www.lejos.org/]

spike11302000, 2016-03-07:
im running ubuntu and i follow this [http://www.ev3dev.org/docs/tutorials/writing-sd-card-image-ubuntu-disk-image-writer/]

dlech, 2016-03-07:
If you plug the SD card into your computer do you see the `EV3DEV_BOOT` and `EV3DEV_ROOT` partitions? 

Do the lights change from red to orange/amber or flash when booting? 

Do you have a serial cable you can use for debugging?

spike11302000, 2016-03-07:
i see rootfs and boot

spike11302000, 2016-03-07:
do i need to rename them?

dlech, 2016-03-07:
no need to rename. and the other questions?

also, make sure nothing is plugged into the EV3 (input ports, output ports and USB).

spike11302000, 2016-03-07:
there is nothing else plugged in

spike11302000, 2016-03-07:
could it be my ev3?

dlech, 2016-03-07:
Doubtful that it is your EV3 if leJOS and the official firmware work.

You still have not answered the other questions.

Do the lights change from red to orange/amber or flash when booting? Do you see the ev3dev logo on the screen?

Do you have a serial cable you can use for debugging on input port 1?

spike11302000, 2016-03-07:
no the light dont change

and what do you mean serial cable, like usb to serial?

spike11302000, 2016-03-07:
it just boot normally

spike11302000, 2016-03-07:
no ev3dev logo

dlech, 2016-03-07:
what (official) firmware do you have in your EV3?

spike11302000, 2016-03-07:
i dont know

i think i need to update

dlech, 2016-03-07:
> and what do you mean serial cable, like usb to serial?

yes

spike11302000, 2016-03-07:
ok then i have a serial cable

dlech, 2016-03-07:
update your firmware first. if you have a 3.3V usb/serial adapter, you can attach it like [this](http://botbench.com/blog/2013/08/15/ev3-creating-console-cable/) and it will hopefully give more info.

dlech, 2016-03-07:
actually a 5V adapter is OK too. just not more than 5V.

spike11302000, 2016-03-07:
ok
i will try that

spike11302000, 2016-03-07:
and i did not know you can get serial data from the ports

spike11302000, 2016-03-07:
can you run a webserver on ev3dev?

spike11302000, 2016-03-07:
just woundering

dlech, 2016-03-07:
yes, you can run a webserver.

spike11302000, 2016-03-07:
can it run php?

dlech, 2016-03-07:
sure. it can run just about anything (that doesn't take up too much memory).

spike11302000, 2016-03-07:
ok
cough cough minecraft server

markosko, 2016-03-07:
not enough processing power ev3 has only 300MHz processor

spike11302000, 2016-03-07:
i know

spike11302000, 2016-03-07:
ok im updating my ev3 now

spike11302000, 2016-03-07:
ok i updated to v1.09h and it still wont boot

spike11302000, 2016-03-07:
and can you boot from the usb port?

dlech, 2016-03-07:
not without compiling your own bootloader

dlech, 2016-03-07:
If leJOS boots from the SD card and ev3dev won't, I'm not sure what could be wrong, sorry.

spike11302000, 2016-03-07:
ok
 im making the serial cable right now so i can see what worng

and can you send me pictures so i can see what on the partions of the sd card

laurensvalk, 2016-03-07:
Not sure it's related, but I occasionally experience that it will not boot into ev3dev. Instead it boots the standard FW from flash. The following works for me:

- Turn the EV3 on.
- In the main menu of the standard firmware, go to the Files tab.
- After some time, you will see the SD card there
- Open that folder. If you can, you know that it's mounted.

Then, shut down the brick through the menu, wait for it to shut down, and launch again. For me this consistently solves my problem, which usually happens the first time after flashing the card. (But not always.)

spike11302000, 2016-03-07:
ok i will try doing that

spike11302000, 2016-03-07:
this mite help

it says cant find boot.src on the sd card

dlech, 2016-03-07:
does `boot.scr` exist in the `EV3DEV_BOOT` partition?

It sounds like the u-boot bootloader is not mounting the sd card properly, which is why @laurensvalk's workaround might help.

does is say anything else besides "cant find uboot.scr"?

spike11302000, 2016-03-07:
[http://i.imgur.com/pmdzRFC.png](url)

spike11302000, 2016-03-07:
http://i.imgur.com/pmdzRFC.png

spike11302000, 2016-03-07:
sorry did the link wrong one the first one

spike11302000, 2016-03-07:
[image]

dlech, 2016-03-07:
You are using the EVB image for BeagleBone Black, so of course it doesn't work on the EV3.

 Try the EV3 image: https://github.com/ev3dev/ev3dev/releases/tag/ev3dev-jessie-2015-12-30

spike11302000, 2016-03-07:
ok..

markosko, 2016-03-07:
i misdownloaded same image at first :D

spike11302000, 2016-03-07:
ok make sence no

spike11302000, 2016-03-07:
thank for the help

closed 2016-03-07 by spike11302000

WasabiFan, 2016-03-07:
@dlech what can be done to name the image files so that they are more obviously different? I imagine that expanding the name in the image file would be a good start ("evb" -> "beaglebone-evb"). Past that, what else could be done? In my mind, the best case would be to have that downloads page on the site that we've wanted so that we can have large title headings.

dlech, 2016-03-07:
normally, all of the images are listed in a single release. however, there was a special release that contains EVB only. this is probably what is throwing people off. I added some capital letters that will hopefully help.