ev3dev/ev3dev #562: avahi-daemon: invalid response packet from host

nickjmeyer, 2016-03-07:
I'm able to get ev3dev loaded on an ev3 without any problem.  However, after the ev3 boots up, about 50% of the cpu is taken for system tasks.

```
top - 18:50:21 up 22 min,  1 user,  load average: 1.31, 1.20, 0.97
Tasks:  59 total,   1 running,  58 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.9 us, 56.3 sy,  0.0 ni, 41.9 id,  0.7 wa,  0.0 hi,  0.2 si,  0.0 st
KiB Mem:     59436 total,    57632 used,     1804 free,     5068 buffers
KiB Swap:    98300 total,     1216 used,    97084 free.    31468 cached Mem
```

I checked `journalctl` and it shows that the avahi-daemon is having some errors.
```
Mar 07 18:51:00 ev3dev avahi-daemon[268]: Invalid response packet from host 10.139.101.26.
Mar 07 18:51:01 ev3dev avahi-daemon[268]: Invalid response packet from host 10.139.101.26.
Mar 07 18:51:02 ev3dev avahi-daemon[268]: Invalid response packet from host 10.139.101.26.
Mar 07 18:51:02 ev3dev avahi-daemon[268]: Invalid response packet from host 10.139.101.26.
```

After googling this error, it appears to be a known bug.  The bug reports say it has been fixed.  Though it appears that the version in ev3dev doesn't have the patch.  Any suggestions for how to proceed?

dlech, 2016-03-07:
Can you provide a link for this bug that you found?

nickjmeyer, 2016-03-07:
Here is a link to the bug post: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=556461

dlech, 2016-03-07:
According to that bug report, the issue was fixed in 0.6.27-2 and we are running 0.6.31-5, so this must be a new/different bug. What else is on your network? Are you using USB, Bluetooth, Wi-Fi?

nickjmeyer, 2016-03-07:
I'm on a campus network, so there are a lot of other machines on the network.  I'm using Wi-Fi.

dlech, 2016-03-07:
Not helpful, but here is a newer unresolved bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=805379

dlech, 2016-03-07:
This looks more useful: https://bugs.launchpad.net/ubuntu/+source/avahi/+bug/1342400

nickjmeyer, 2016-03-07:
Well, it sounds like this isn't an EV3 issue then.  I'll keep an eye on that bug report.

nickjmeyer, 2016-03-07:
Interesting.  Looks like they're saying it is fixed in 0.6.32

dlech, 2016-03-07:
yeah, debian has a 0.6.32-rc package in testing. you could try installing that and see what happens.

https://packages.qa.debian.org/a/avahi.html

nickjmeyer, 2016-03-07:
I added `APT:Default-Release "Jessie";` to `/etc/apt/conf` and `deb http://ftp.debian.org/debian sid main contrib non-free` to `/etc/apt/sources.list`.

Then ran `apt-get update && apt-get install avahi-daemon`.

Not sure if that is the correct way to install the updated package.  Hopefully I didn't break something else...

I reloaded the image onto the ev3 and things are much better.  I still see the same error message in the journal.  However, the frequency seems to be much less.  The system cpu usage is only ~5% now.

dlech, 2016-03-07:
cool. I'll look at backporting this to jessie or including the sid version in the ev3dev.org repository so that it is included by default in the next image release.

nickjmeyer, 2016-03-07:
Is it going to break anything if I leave ` deb http://ftp.debian.org/debian sid main contrib non-free` in `/etc/apt/sources.list`?  I'm assuming as long as I don't do `apt-get upgrade` then it is okay?

dlech, 2016-03-07:
Yes, it's ok as long as you don't upgrade all of your packages. You could disable `sid` in your apt sources to be safe.

nickjmeyer, 2016-03-07:
Awesome.  Thank you for all of your help.  Very much appreciated.

dlech, 2016-04-11:
We are now shipping `avahi 0.6.32-1ev3dev1` in the ev3dev.org repository, so no need for installing the package from sid.

https://github.com/ev3dev/avahi

closed 2016-04-11 by dlech