ev3dev/ev3dev #563: Support for raspberry pi 3?

JorgePe, 2016-03-07:
Hi.
I got my first RPi3. I thought the RPi3 is just an extended version of the RPi2 but it seems I'm wrong:

Tried ev3dev for rpi3 (latest version, rpi2-ev3dev-jessie-2015-12-30.img.xz). Never shows any kind of text, just a screen with lot of colors.
Tried raspbian (2016-02-26-raspbian-jessie-lite.img) and it boots so hardware is fine.
Then tried ev3dev with a RPi2 and it also boots.

Any kind of configuration I can do when running raspbian that can force some kind of compatibility mode or I just/must need to wait for a rpi3 version of evdev?

dlech, 2016-03-07:
No idea. I don't know that anyone has got RPi3 working with something other than the official raspbian yet.

dlech, 2016-03-07:
I did read that the RPi 3 uses significantly more power, so it does not seem like the best choice for battery powered applications.

JorgePe, 2016-03-08:
Yes, almost 50% more current. But I was considering using it with a PSU, over a table, with a BrickPi and a camera, and some image processing.
Will try to find more info.

JorgePe, 2016-03-10:
OK, I got it. There are other people complaining (other distros).
I don't know exactly what is the problem but this makes ev3dev to boot on a rpi3:

I made a microSD card with "rpi2-ev3dev-jessie-2015-12-30.img.xz" and used it on a rpi2, configured, upgraded and used it just a lit bit.
Then I extracted all files from boot partition of raspbian (2016-02-26) to the ev3dev boot partition (only ketp cmdline.txt)
It booted and I could login (locally).

It's an ugly and clumsy hack but at least I know there is hope for ev3dev on Raspberry Pi 3.

jabrena, 2016-03-10:
Maybe using a external USB battery could be enough to solve energy issues.
https://www.powertraveller.com/au/shop/portable-chargers/professional/powergorilla/

dlech, 2016-03-10:
Actually how about just `ls /boot/flash`.

JorgePe, 2016-03-11:
cat /proc/cpuinfo

```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a02082
Serial		: 00000000b6f07128
```

cat /proc/device-tree/model; echo
```
Raspberry Pi 3 Model B Rev 1.2
```

ls /boot/flash -l
```
total 23762
-rwxr-xr-x 1 root root   10841 Feb 25  2016 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   11120 Feb 25  2016 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   10871 Feb 25  2016 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   12108 Feb 25  2016 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   11555 Dec 30  2015 bcm2709-rpi-2-b.dtb.bak
-rwxr-xr-x 1 root root   12575 Feb 25  2016 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   17920 Feb  9  2016 bootcode.bin
-rwxr-xr-x 1 root root      97 Dec 30  2015 cmdline.txt
-rwxr-xr-x 1 root root    1635 Feb 26  2016 config.txt
-rwxr-xr-x 1 root root   18693 Aug 21  2015 COPYING.linux
-rwxr-xr-x 1 root root    2505 Feb 25  2016 fixup_cd.dat
-rwxr-xr-x 1 root root    6481 Feb 25  2016 fixup.dat
-rwxr-xr-x 1 root root    9722 Feb 25  2016 fixup_db.dat
-rwxr-xr-x 1 root root    9724 Feb 25  2016 fixup_x.dat
-rwxr-xr-x 1 root root     103 Feb 26  2016 issue.txt
-rwxr-xr-x 1 root root 4046732 Feb 25  2016 kernel7.img
-rwxr-xr-x 1 root root 4049396 Dec 30  2015 kernel7.img.bak
-rwxr-xr-x 1 root root 3963140 Feb 25  2016 kernel.img
-rwxr-xr-x 1 root root    1494 Nov 18  2015 LICENCE.broadcom
-rwxr-xr-x 1 root root   18974 Feb 26  2016 LICENSE.oracle
drwxr-xr-x 2 root root    6144 Mar 10  2016 overlays
-rwxr-xr-x 1 root root     136 Feb 26  2016 _cmdline.txt
-rwxr-xr-x 1 root root  612472 Feb 25  2016 start_cd.elf
-rwxr-xr-x 1 root root 4888200 Feb 25  2016 start_db.elf
-rwxr-xr-x 1 root root 2739672 Feb 25  2016 start.elf
-rwxr-xr-x 1 root root 3840328 Feb 25  2016 start_x.elf
```

dlech, 2016-03-11:
https://github.com/raspberrypi/linux/commit/866cf9417ee07f6695f28af068a518b71df5916c

dlech, 2016-03-11:
Have to wait and see if it is legal to redistribute Bluetooth and Wi-Fi firmware:

https://github.com/raspberrypi/linux/issues/1325#issuecomment-195305907

dlech, 2016-04-11:
Updated `rpi-firmware` package for RPi3 (this is the boot partition stuff)

https://github.com/ev3dev/rpi-firmware/commit/ab592426600b23c65b0aa13164b1e6dd24db495c

dlech, 2016-04-11:
Kernel with RPi 3 support is released: 

https://github.com/ev3dev/linux-latest-ev3dev-rpi2/commit/a394e69b7e0d7e2dd5d985cb5a13c9b7de614a40

dlech, 2016-04-11:
At this point, you should be able to run

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade
    sudo reboot

on your hacked RPi 3 image and have a working system.

There might be issues with Bluetooth and BrickPi since as I understand it, by default the RPi 3 uses the same UART for Bluetooth that is used for communication to the BrickPi. If this is the case, please open a new issue for that topic.

JorgePe, 2016-04-11:
Wonderful!
It worked, just had to add an USB Wi-Fi dongle to pass the network configuration loop.
It recognizes ethernet card but doesn't recognize internal Wi-Fi nor internal Bluetooth (but works OOB with my USB BT4.0 dongle).
Will test BrickPi later.
Thanks a lot, ev3dev runs on Raspberry Pi 3. Hope in a near future it works wih Wi-Fi and BT also.

closed 2016-04-11 by JorgePe

dlech, 2016-04-11:
Please open new issues for RPi 3 built-in Wi-Fi and Bluetooth.