ev3dev/ev3dev #564: BrickPi firmware 2.8

danleavitt0, 2016-03-09:
@dlech BrickPi just added an update firmware to their github repository. Do you know if this addresses some of the issues with uART sensors?

dlech, 2016-03-09:
Looks like 2.5 rather than 2.8.

@DexterInd, can you fill us in on what has changed?

DexterInd, 2016-03-10:
@dlech and @danleavitt0 , no huge changes here, we were just catching up on publishing the code we're shipping with our hardware lately.  The big changes are in how motors are handled on the very low level.  In the BrickPi+ we switched out to a more robust motor chip, which had slightly different wiring than the BrickPi Advanced.

DexterInd, 2016-03-10:
What UART issues are we seeing with the BrickPi?

dlech, 2016-03-10:
So firmware 2.5 is for BrickPi+ only because of the wiring differences?

UART sensor issue: https://github.com/DexterInd/BrickPi/issues/24

joshrtay, 2016-03-10:
@DexterInd a big +1 for fixing all of @dlech issues. 

brickPi is way faster than the ev3 (reads and writes average 1.25ms vs 4.5ms on ev3), so we would love to use the brickPi with ev3dev, but right now it's too buggy.

DexterInd, 2016-03-11:
@dlech yes, I believe so.  

@joshrtay Thanks for the feedback.

joshrtay, 2016-03-11:
@dlech Can you confirm that 2.5 fixes this issues?

dlech, 2016-03-11:
I don't have a BrickPi+, but it sounds like no issues have been fixed. There where just pin changes for the new motor controller.

dlech, 2016-03-14:
Questions seem to be answered and there is nothing to do, so closing.

closed 2016-03-14 by dlech