ev3dev/ev3dev #567: EV3 nxt light sensor shows always 0 after installing patched kernel

markosko, 2016-03-12:
After i installed patched kernel which repaired reading value of compass sensor, nxt light sensor shows always 0. It's detected in list of devices can change mode but value is always 0 dont know whats problem

harmooni, 2016-03-12:
Yes sorry, that's my fault. Because I put brackets around 
```
(mode_info->si_max - mode_info->si_min)
			/ (mode_info->raw_max - mode_info->raw_min)
```
 [(you can find the corresponding lines of code here)](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c#L400),
the value of the light sensor is always multiplied by 0.
We should remove the brackets again (@dlech). It should still work with the HT angle sensor because of the altered if statement. I can test that on monday.
But it would be good to make sure that no accidental overflow happens there again.

markosko, 2016-03-12:
and whats this line for which sensor ? or can u tell me whats was original ?

harmooni, 2016-03-13:
This is not part of a definition for a specifc sensor. It belongs to the general handling of sensor data.
The original code was:
```
if (mode_info->raw_min != mode_info->raw_max) {
                 (mode_info->si_max - mode_info->si_min)
 	        / (mode_info->raw_max - mode_info->raw_min)
}
```
I altered it to [ (lego_sensor_class.c - line 397 and the following) ](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/lego_sensor_class.c#L397)
```
if (mode_info->raw_min != mode_info->si_min 
					|| mode_info->raw_max != mode_info->si_max) {						
		*value = (*value - mode_info->raw_min)
			* ((mode_info->si_max - mode_info->si_min)
			/ (mode_info->raw_max - mode_info->raw_min))
			+ mode_info->si_min;
	}
```
The brackets around 
(mode_info->si_max - mode_info->si_min)  / (mode_info->raw_max - mode_info->raw_min)
cause the problem with the light sensor.

markosko, 2016-03-13:
Yes removing brackets solved problem thanks :)

closed 2016-03-13 by markosko