ev3dev/ev3dev #568: First connection with BrickPi

jabrena, 2016-03-13:
Hi,

I have created a SD Card for RaspberryPi2 and I have updated config.txt for Brickpi and I have connected the raspi with brickpi to the router using a RJ45 wire and I turned on.

But checking what host are connected in the network, I didn't see brickpi.

I checked the SD Card with PiStorms updating config.txt and I discarted that exist a problem with SD.
How to make a ssh connection with BrickPi?

I am a bit puzzle with this comment:
http://www.ev3dev.org/docs/getting-started/

``` bash
For Raspberry Pi with no display, you must use the wired Ethernet port to connect for the first time. You can setup additional connections using the connmanctl command.
```
How to connect brickpi using ssh the first time?

Cheers

Juan Antonio

dlech, 2016-03-13:
`ssh robot@ev3dev` or `ssh robot@ev3dev.local` ?

jabrena, 2016-03-13:
Hi mate,

yes, the problem was mine, I had to uncomment both lines for brickpi.

``` bash
robot@ev3dev:/sys/class/tty/ttyAMA0/device/of_node$ cat /sys/class/tty/ttyAMA0/device/of_node/brickpi,status
okay
```

Many thanks @dlech 

Cheers

closed 2016-03-13 by jabrena