ev3dev/ev3dev #570: ev3dev-config: selecting "Hardware Configuration" does nothing on EVB

dlech, 2016-03-14:
Haven't look into it yet, but selecting "Hardware Configuration" flashes to something else for a split second then goes back to the main menu. Seems like it ought to show a blank page that says "nothing to do" if there is really nothing to do.

WasabiFan, 2016-03-14:
That sounds like an error is being thrown. I don't see anything wrong from a quick read-through, but I'd guess that one of the first two calls in [this function](https://github.com/ev3dev/ev3dev-tools/blob/ev3dev-jessie/source/scripts/ev3dev-config.sh#L264) is failing.

WasabiFan, 2016-04-04:
I have opened a PR to resolve this issue, can you try from that PR branch and see if the problem is resolved? I was able to repro by dismantling auto-detection so that it detected the EV3 as a Beagle device; from there I just fixed the most obvious problem and it fixed it on my end.

dlech, 2016-06-17:
fix has been released in `ev3dev-tools 1.2.0`

closed 2016-06-17 by dlech