ev3dev/ev3dev #571: still out of memory when building Erlang on EV3 after update zraminit 3.5ev3dev2

han88, 2016-03-14:
Still running out of memory after i increased the swap size to 96MB.. any help?! thx

CC	/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o

cc1: out of memory allocating 11582624 bytes after a total of 17690624 bytes
pcre/pcre.mk:64: recipe for target '/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o' failed
make[3]: *** [/home/robot/otp_src_18.2.1/erts/emulator/pcre/obj/armv5tejl-unknown-linux-gnueabi/opt/pcre_exec.o] Error 1

dlech, 2016-03-14:
Don't compile on the brick. [Use brickstrap](http://www.ev3dev.org/docs/tutorials/using-brickstrap-to-cross-compile/).

dlech, 2016-03-14:
Or create a very large swap file on your SD card (not recommended).

https://github.com/ev3dev/ev3dev/issues/520#issue-128399751

han88, 2016-03-14:
using brickstrap i get this error:
=== Entering application hipe
make[3]: Entering directory '/home/otp_src_18.2.1/lib/hipe/rtl'
make[4]: Entering directory '/home/otp_src_18.2.1/lib/hipe/main'
 VSN	hipe.hrl
make[4]: Leaving directory '/home/otp_src_18.2.1/lib/hipe/main'
 ERLC	../ebin/hipe_rtl.beam
/build/qemu-_D3HGx/qemu-2.0.0+dfsg/tcg/tcg.c:179: tcg fatal error

dlech, 2016-03-14:
What is your host OS? This looks like a bug in qemu. [Here is the source code tht the errror points to.](https://sources.debian.net/src/qemu/2.0.0%2Bdfsg-4~bpo70%2B1/tcg/tcg.c/#L179). No idea what it means though or what you can do about it.

brickstrap runs on qemu, so sometimes things just might not work if qemu can't handle it. If you have an RPi or other ARM device with more memory, it would be faster to compile there (should work using multiarch to install armel stuff), but you might be resigned to building on the EV3 in this case. See the link in my second command on how to create a nice big swap partition.

han88, 2016-03-14:
ubuntu 14.04 lts... 
i set swap file size to 512mb by:
fallocate -l 512M /swapfile
chmod 600 /swapfile
mkswap /swapfile
swapon /swapfile
...but still getting the out of memory error, is that too little?!

dlech, 2016-03-14:
What is `cat /proc/swaps` on the EV3? Open a second ssh terminal and do this while compiling. 512M should be plenty you would think.

han88, 2016-03-14:
so that looks strange... this was the highest amount of memory use
robot@ev3dev:~$ cat /proc/swaps
Filename				Type		Size	Used	Priority
/dev/zram0                              partition	98300	98300	16383
/swapfile                               file		524284	18432	-1

... obviously there is no full use of the memory?!
jfcloutier wrote he could build erlang after your upgrade zraminit 3.5ev3dev2

dlech, 2016-03-14:
Hmm... the zram swap is actually just compressed ram, so since it is full, it could be filling up your actual ram. Try turning off the zram swap and just using your swapfile.

    sudo systemctl stop zram_swap.service

dlech, 2016-03-14:
Maybe your are using a different configuration for compiling and it uses more memory.

han88, 2016-03-14:
yeah!! turned off zram swap... it works!! thx a lot

closed 2016-03-14 by han88