ev3dev/ev3dev #572: Bluetooth does not start

laurensvalk, 2016-03-14:
After launch, Bluetooth often does not come up, and it fails to turn on, showing '??':

- [Screen before turning BT on] (https://cloud.githubusercontent.com/assets/12326241/13756808/f7f22a34-ea21-11e5-9fef-6e1bd1b12d21.jpg)
- [Screen after turning BT on] (https://cloud.githubusercontent.com/assets/12326241/13756914/64af695c-ea22-11e5-8006-d4cc6d43a34b.jpg)

Workaround: Attempt to turn bluetooth on; see '???' on the screen; reboot.

Essentially two boots are typically required to turn on BT.

The following is going to sound vague, but somehow it seems that since ev3dev-jessie-2015-12-30, bluetooth connections are very hard to (re-)establish, frequently giving a GDBus error.

dlech, 2016-03-14:
This has always been a problem with jessie. Since we are using systemd for init, it does many things at once and the hciattach command times out. I haven't figured out a good way to fix it yet. You can tell when this happens by running `sudo systemctl -l status ev3-bluetooth.service`. This usually only happens to me on the very first boot with a new SD card image because we are generating ssh keys and a few extra things so the system is extra busy. After that, it is generally fine. But it sounds like you are saying that this happens randomly on subsequent boots?

Is re-establishing connections a separate issue from the no Bluetooth at all ("???") ? If so, please open a second issue for this.

laurensvalk, 2016-03-14:
Indeed, this happens more often, and I haven't found a pattern yet.

I'll open another issue for other BT issues.

dlech, 2016-03-14:
Next time it happens, run `sudo systemctl -l status ev3-bluetooth.service` before you reboot and we will see if it is the same error I have seen.

antonvh, 2016-03-16:
Same issue here. Rebooting the mac AND the Ev3 usually does the trick. I suspect it has to do with computers going to sleep while bluetooth connections are open. I disabled 'connect automatically' in brickman for a more reliable boot of the ev3 brick: sometimes it doesn't boot at all, sometimes it boots in bluetooth "???" mode. @laurensvalk and I are working together on project anyway so we'll try to dig up the source of this.

dlech, 2016-03-17:
When bluetooth does not start (indicated by "???" on the EV3), please post the output of `journalctl -b`. Please use a Gist or pastebin since it will be quite long.

dlech, 2016-04-12:
One thing I have noticed is that this happens more often if many EV3/UART sensors are attached. In particular the Gyro sensor which seems to use high CPU because of high data rates. In the `3.16.7-ckt26-10-ev3dev-ev3` kernel, I have offloaded EV3/UART sensor data processing to a separate kernel thread. I don't know if this will make a difference or not, but pay attention when you upgrade to this kernel and see if you can notice a difference.

laurensvalk, 2016-04-30:
> When bluetooth does not start (indicated by "???" on the EV3), please post the output of sudo journalctl -b.

It's [here] (https://dl.dropboxusercontent.com/u/19872448/journaloutput).

dlech, 2016-04-30:
Just picking out the relevant parts for later comparison:

```
Jan 01 01:00:24 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
...
Jan 01 01:00:26 ev3dev kernel: Bluetooth: Core ver 2.19
Jan 01 01:00:26 ev3dev kernel: NET: Registered protocol family 31
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCI device and connection manager initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCI socket layer initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: L2CAP socket layer initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: SCO socket layer initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCI UART driver ver 2.2
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCI H4 protocol initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCI BCSP protocol initialized
Jan 01 01:00:26 ev3dev kernel: Bluetooth: HCILL protocol initialized
...
Jan 01 01:00:27 ev3dev kernel: Bluetooth: hci0 sending frame failed
...
Jan 01 01:00:28 ev3dev hciattach[259]: Cannot send hci command to socket: Connection timed out
Jan 01 01:00:28 ev3dev hciattach[259]: Can't initialize device: Connection timed out
...
Jan 01 01:00:29 ev3dev systemd[1]: ev3-bluetooth.service: control process exited, code=exited status=1
Jan 01 01:00:29 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Jan 01 01:00:29 ev3dev systemd[1]: Unit ev3-bluetooth.service entered failed state.
```

dlech, 2016-05-07:
Here's one from me. Very similar, but I don't have `Bluetooth: hci0 sending frame failed`.

http://paste.debian.net/682670/

```
...
May 07 03:45:03 ev3dev kernel: Linux version 4.4.9-dlech-ev3+ (david@freyr) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11) ) #61 PREEMPT Fri May 6 22:34:52 CDT 2016
...

May 07 03:45:44 ev3dev systemd[1]: Reached target Basic System.
May 07 03:45:45 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 UART Sensors...
May 07 03:45:45 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 UART Sensors...
May 07 03:45:45 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
May 07 03:45:45 ev3dev systemd[1]: Starting Login Service...
...
May 07 03:45:46 ev3dev systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
May 07 03:45:46 ev3dev systemd[1]: Starting D-Bus System Message Bus...
May 07 03:45:47 ev3dev systemd[1]: Started D-Bus System Message Bus.
May 07 03:45:49 ev3dev kernel: Registered EV3 UART sensor line discipline. (29)
May 07 03:45:50 ev3dev kernel: lego-sensor sensor0: Registered 'lego-ev3-gyro' on 'in2'.
May 07 03:45:49 ev3dev avahi-daemon[274]: Found user 'avahi' (UID 103) and group 'avahi' (GID 105).
May 07 03:45:49 ev3dev avahi-daemon[274]: Successfully dropped root privileges.
May 07 03:45:50 ev3dev avahi-daemon[274]: avahi-daemon 0.6.32 starting up.
May 07 03:45:50 ev3dev hciattach[266]: Found a Texas Instruments' chip!
May 07 03:45:50 ev3dev hciattach[266]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.2.31.bts
May 07 03:45:50 ev3dev hciattach[266]: Loaded BTS script version 1
May 07 03:45:50 ev3dev hciattach[266]: texas: changing baud rate to 2000000, flow control to 1
May 07 03:45:51 ev3dev hciattach[266]: Cannot send hci command to socket: Connection timed out
May 07 03:45:51 ev3dev hciattach[266]: Can't initialize device: Connection timed out
May 07 03:45:51 ev3dev kernel: lego-sensor sensor1: Registered 'lego-ev3-color' on 'in4'.
May 07 03:45:51 ev3dev kernel: lego-sensor sensor2: Registered 'lego-ev3-us' on 'in3'.
May 07 03:45:52 ev3dev avahi-daemon[274]: Successfully called chroot().
May 07 03:45:52 ev3dev avahi-daemon[274]: Successfully dropped remaining capabilities.
May 07 03:45:52 ev3dev avahi-daemon[291]: chroot.c: open() failed: No such file or directory
May 07 03:45:52 ev3dev avahi-daemon[274]: Failed to open /etc/resolv.conf: Invalid argument
May 07 03:45:52 ev3dev avahi-daemon[274]: No service file found in /etc/avahi/services.
May 07 03:45:52 ev3dev dbus[278]: [system] Successfully activated service 'org.freedesktop.systemd1'
May 07 03:45:52 ev3dev avahi-daemon[274]: socket() failed: Address family not supported by protocol
May 07 03:45:52 ev3dev avahi-daemon[274]: Failed to create IPv6 socket, proceeding in IPv4 only mode
May 07 03:45:53 ev3dev avahi-daemon[274]: socket() failed: Address family not supported by protocol
May 07 03:45:53 ev3dev avahi-daemon[274]: Network interface enumeration completed.
May 07 03:45:53 ev3dev avahi-daemon[274]: Server startup complete. Host name is ev3dev.local. Local service cookie is 2844453812.
May 07 03:45:52 ev3dev systemd[1]: Started Avahi mDNS/DNS-SD Stack.
May 07 03:45:52 ev3dev systemd[1]: Starting Permit User Sessions...
May 07 03:45:53 ev3dev systemd[1]: Starting Turn on global VT cursor...
May 07 03:45:53 ev3dev systemd[1]: Started getty on tty2-tty6 if dbus and logind are not available.
May 07 03:45:53 ev3dev systemd[1]: Starting USB Gadget for LEGO MINDSTORMS EV3 hardware...
May 07 03:45:54 ev3dev systemd[1]: Started Restore Sound Card State.
May 07 03:45:55 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
May 07 03:45:55 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
May 07 03:45:55 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
May 07 03:45:55 ev3dev systemd[1]: ev3-bluetooth.service: control process exited, code=exited status=1
May 07 03:45:56 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
May 07 03:45:56 ev3dev systemd[1]: Unit ev3-bluetooth.service entered failed state.
May 07 03:45:56 ev3dev systemd[1]: Started Permit User Sessions.
May 07 03:45:56 ev3dev systemd[1]: Started Turn on global VT cursor.
```

dlech, 2016-05-07:
Interesting. One time, I had a sensor fail instead of Bluetooth.

```

May 07 04:15:03 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
May 07 04:15:03 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 UART Sensors.
May 07 04:15:03 ev3dev systemd[1]: ev3-uart@in3.service: control process exited, code=exited status=1
May 07 04:15:03 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 UART Sensors.
May 07 04:15:03 ev3dev systemd[1]: Unit ev3-uart@in3.service entered failed state.
May 07 04:15:04 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 built-in Bluetooth.
```

dlech, 2016-05-08:
Is the situation improved with <http://www.ev3dev.org/news/2016/05/08/Kernel-Release-Cycle-11/>?

markyi370, 2016-05-10:
Can we move the bluetooth inilization earlier in the boot process?

dlech, 2016-05-10:
Possibly. You could try adjusting the dependencies in [ev3-bluetooth.service](https://github.com/ev3dev/ev3-systemd/blob/ev3dev-jessie/systemd/ev3-bluetooth.service). [Docs](https://manpages.debian.org/cgi-bin/man.cgi?query=systemd.unit).

But first, please try with the `4.4.9-11-ev3dev-ev3` kernel and see if there is any difference.

dlech, 2016-05-10:
Also, [legoev3_bluetooth](https://github.com/ev3dev/lego-linux-drivers/blob/master/ev3/legoev3_bluetooth.c) is probably being loaded as a module. If we compiled it into the kernel, that would make it available earlier.

johngpayne, 2016-07-11:
As requested, moving my comment to this bug as similar issue. I'm getting roughly the same symptoms - the same ??? in the bluetooth menus as above. The key information in sudo journalctl looks to be:

Jan 01 00:01:00 ev3dev brickman[321]: ** (process:321): CRITICAL **: bluez5_manager_get_devices: assertion 'self != NULL' failed
Jan 01 00:01:00 ev3dev dbus[272]: [system] Failed to activate service 'org.bluez': timed out
Jan 01 00:01:22 ev3dev systemd[1]: ev3-bluetooth.service: control process exited, code=exited status=1
Jan 01 00:01:22 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Jan 01 00:01:22 ev3dev systemd[1]: Unit ev3-bluetooth.service entered failed state.
Jan 01 00:01:22 ev3dev systemd[1]: Starting Connection service...
Jan 01 00:01:22 ev3dev systemd[1]: Starting Bluetooth service...
Jan 01 00:01:23 ev3dev bluetoothd[451]: Bluetooth daemon 5.23
Jan 01 00:01:23 ev3dev systemd[1]: Started Bluetooth service.
Jan 01 00:01:24 ev3dev kernel: Bluetooth: Core ver 2.19

I'm on kernel 3.16.7-ckt21-9-ev3dev-ev3 (willing to try later - seen mention of 4.4.9 above) but 3.16.7 seems to be the latest that apt-get will get for me?)

dlech, 2016-07-11:
You have to run `sudo apt-get dist upgrade to upgrade the kernel. However, I don't expect that to make any difference. As I mentioned in the duplicate issue, unplugging sensors and upgrading flash-kernel may help.

Although, if bluetooth is failing to start every time and not just sometimes, there may be something else going on.

dlech, 2016-07-24:
I think we have this mostly sorted out. Please try with the latest nightly image build.

 [ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

johngpayne, 2016-07-25:
Bluetooth seems to work for me with this image. Thanks!

dlech, 2016-10-11:
Please re-open if this is still an issue.

closed 2016-10-11 by dlech

antonvh, 2016-10-12:
It is in the latest image on ev3dev.org (dec 2015), but it's fixed in the
latest nightly build.