ev3dev/ev3dev #574: RTC_HCTOSYS [=n]

dlech, 2016-03-15:
Since the EV3 doesn't have a battery for the RTC, this is probably a good idea in combination with #573.

```
CONFIG_RTC_HCTOSYS:                                                                                       │  
  │                                                                                                           │  
  │ If you say yes here, the system time (wall clock) will be set using                                       │  
  │ the value read from a specified RTC device. This is useful to avoid                                       │  
  │ unnecessary fsck runs at boot time, and to network better.                                                │  
  │                                                                                                           │  
  │ Symbol: RTC_HCTOSYS [=n]                                                                                  │  
  │ Type  : boolean                                                                                           │  
  │ Prompt: Set system time from RTC on startup and resume                                                    │  
  │   Location:                                                                                               │  
  │     -> Device Drivers                                                                                     │  
  │       -> Real Time Clock (RTC_CLASS [=y])                                                                 │  
  │   Defined at drivers/rtc/Kconfig:20                                                                       │  
  │   Depends on: RTC_CLASS [=y]       
```

jabrena, 2016-03-18:
It is pretty interesting. 
What is your idea if the brick doesn't have an Internet connection?

dlech, 2016-03-19:
I haven't looked into this enough that I have an opinion at this time.

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel. Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech