ev3dev/ev3dev #578: Question: Using a different EV3, unable to connect

demar42, 2016-03-21:
Hi there,

I have set up an EV3 with ev3dev to be able to communicate with my computer (I'm using windows) via putty and winSCP by following the instructions on the ev3dev website. However, due to certain circumstances, I have had to change ev3 brick. When I plugged the SD card into the new brick and connected it to my computer, under network adapter settings its type is "EV3+ev3dev #3". It used to be just "EV3+ev3dev" and I could connect to it with hostname "ev3dev". However, now when I use "ev3dev" or "ev3dev #3" or its IP address, I cannot connect. When using "ev3dev" or "ev3dev #3" putty tells me "no hostname", and when I use the IP address it opens a terminal window but says the connection timed out. I have tried re-flashing the SD card but to no avail.

Thanks in advance for your help!

laurensvalk, 2016-03-21:
This issue (#527) might be similar. You'll find some answers from the devs there.

dlech, 2016-03-21:
The MAC address for USB (and Bluetooth) is tied to the physical hardware, not the SD card. This is why it shows up as a different connection when you put the SD card in a different EV3 and also why there is not change when you re-flash the SD card and use it in the same EV3.

Are you using a static IP address on the EV3? If so, that could be part of the problem. Windows keeps an arp cache that associates MAC address to IP addresses, so if both EV3s are using the same IP address, Windows could be trying to send packets to the wrong EV3.

Also you must totally disable Internet Connection Sharing and then set it up again when you change EV3s. 

And the Windows NETBIOS table probably still has the name "ev3dev" associated with the first EV3 too.

Here is a link on diagnosing TCP/IP connection issues on Windows: <https://technet.microsoft.com/en-us/library/cc739601%28v=ws.10%29.aspx>.

TL;DR;

In a windows command prompt, run

    nbtstat -n
    arp -a

If the NETBIOS table is wrong, run 

    nbtstat -RR

If the ARP info is wrong, use 

    arp -d <ip-address>

demar42, 2016-03-23:
Hi there,

Thank you very much for your help, I love how this community is so active!

I tried what you suggested above, but it didn't seem to work. After a bit of researching and trial and error, I figured out that to remove the # number after the "ev3dev", I had to go into regedit and delete the "EV3+ev3dev" registry key at `HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Control\Network\{long hex number}\Descriptions` then remove the device from devices and printers (right-click > remove device). When I plugged it back in, it installed the device driver software and then it came up as "ev3dev", which was exactly what I wanted.

However, after that, I ran into the issue of not being able to connect with ssh. Both winSCP and PuTTY couldn't connect, giving the error "Software caused connection abort". I found out that in the process of re-flashing the SD card to fix the error above, something had gone wrong and the ssh was not working. I re-flashed the SD again, and now it works!

Thank you dlech for your wonderful help, this issue to now resolved.

closed 2016-03-23 by demar42