ev3dev/ev3dev #579: Question: "Software caused connection abort" on both winSCP and PuTTY

demar42, 2016-03-22:
Hi there,

I have been trying to connect to my ev3 which is running ev3dev. I am on a windows machine and use either putty or winscp to connect. However, when I try to connect with either winscp or putty using hostname ev3dev, it fails, saying "Software caused connection abort". I have added exceptions in the firewall for winscp but it still doesn't work.

Thanks in advance for your help!

dlech, 2016-03-22:
How are you connecting? USB, Bluetooth, Wi-Fi? Have you tried connecting using the IP address instead of `ev3dev`?

dlech, 2016-03-22:
Is this related to #578?

demar42, 2016-03-22:
Yes, this is related to #578, in that the new ev3 brick I managed to set up is now not able to be connected. 
I am currently connecting over USB, but I will try the IP address connection as suggested when possible.
Thanks for your help!

dlech, 2016-03-22:
Try the things I suggested in #578 and report back there. No sense in having to open issues for the same problem.

closed 2016-03-22 by dlech