ev3dev/ev3dev #580: What is the minimum voltage for BrickPi?

jabrena, 2016-03-23:
Hi,

I am testing BrickPi and I would like to know what is the minimum voltage to shutdown a brickpi unit:

``` bash
robot@ev3dev:/sys/class/power_supply/brickpi-battery$ cat voltage_now 
10280500
```

Juan Antonio

dlech, 2016-03-23:
I don't know, use a variable power supply and find out experimentally, read the data sheet on the power regulator chip or ask @DexterInd.

DexterInd, 2016-03-24:
Probably should not let the input voltage drop below 7 Volts (IE Battery voltage).  The chip we use for voltage conversion down to 5V can have a 1-1.5 (depending on load) drop out V, so you need to make sure that it's above that or your Raspberry Pi might start behaving erratically.

WasabiFan, 2016-03-24:
@jabrena 
> ...and I would like to know what is the minimum voltage to shutdown a brickpi unit:

Note that, as @DexterInd mentioned, you likely won't actually see the device shutdown. Instead, the system will behave in unexpected ways; chips might restart/reset, motors and sensors won't respond correctly, and it's _possible_ that you could damage your Pi or other devices (although that is very unlikely for under-voltage -- it's going over that is dangerous). Just be aware that it might not be obvious that the BrickPi/RasPi isn't happy.

DexterInd, 2016-03-24:
Yes, you'll see funny behavior.  Most likely behavior will be that networking shuts down.  Typically on lower powers, the Pi starts to shut down wifi and networking first.  That's just what we've observed though, you might see something totally different.

jabrena, 2016-03-24:
It could be interesting some development to monitor battery to advise the robot to shutdown to prevent the data lost or some problem with the sdcard.

``` bash
robot@ev3dev:/sys/class/power_supply/brickpi-battery$ cat voltage_now 
7586300
``` 

Near to the battery threshold!!! :D

DexterInd, 2016-03-24:
Hey Juan, it could cause problems and it would be great to use the battery monitor to prevent this.  How would you do it thought without interrupting power users?  And what would you recommend doing: shutting the Pi down?

dlech, 2016-03-24:
Since BrickPi does not have a display or speaker not notify users, we are left with the LEDs and the terminal. We could blink the LEDs in a certain pattern when the battery is getting low. However if the user program is using LEDs, this could be missed.

Another option is to use the [wall](http://manpages.debian.org/cgi-bin/man.cgi?query=wall) command to broadcast a message when the battery is getting low. Again, this could be missed if the user does not have a terminal open to the BrickPi.

If the battery gets too low, I think shutting down is the right thing to do.

DexterInd, 2016-03-24:
Maybe, on a new version, we should put a LED warning or a low-battery led?

dlech, 2016-03-24:
I think a low battery LED would be a nice feature. It should only turn on when battery is low so it normally does not consume power and it will have a dedicated purpose, so it won't be busy doing something else.

jabrena, 2016-03-24:
Hi, the idea from @dlech could be a good beginning because it is easy to implement in a user project. The feature that @dlech says about a led could be a improvement for the product. Another solution with more work could be connect a bluetooth speaker to play a simple beep or a expression with eSpeak in a user project.

Many thanks for the data: 7 Volts

Cheers

closed 2017-07-25 by dlech