ev3dev/ev3dev #582: SSH connection user and pwd not working

BorisFaz, 2016-03-23:
Hi
I'm a new user.
I Set up ev3 and I'm using the image loaded into the SD card. I'connected the EV3 to the netrwok and it has a valid IP address. Now, from Windows with putty I'm trying to connect to the EV3 but user and pwd are not valid. Can someone tell me what I'm missing?
My final goal is to compile a simple program and execute it.
Thanks in advance.

WasabiFan, 2016-03-23:
What user name and password are you using, and what specific image did you use to image the SD card? The default user and password are `robot` and `maker`, so make sure you are trying those.

dlech, 2016-03-23:
Also, please fill out as many questions as possible in the issue template when posting an issue. It will give us a more complete idea of what you have tried already so that we can help you better.

```
System information:

    I am using: EV3

    I installed ev3dev using this image file: ???

    My kernel version is: ???

    My host computer is running: Windows 7/8/10?

    I am connecting using: ???

About my issue:

What led up to the situation?

I Set up ev3 and I'm using the image loaded into the SD card. I connected the EV3 to the network and it has a valid IP address.

What exactly did you do (or not do) that was effective (or ineffective)?

Now, from Windows with putty I'm trying to connect to the EV3 but user and pwd are not valid.

What was the outcome of this action?

What outcome did you expect instead?

```

Some missing relevant information is how are you connecting (USB, Bluetooth, Wi-Fi)?

What is the exact error message from PuTTY?

dlech, 2016-03-23:
Every once in a while, the SSH keys are not setup correctly on the first boot and you have to re-flash the SD card, so that is something else you can try.

BorisFaz, 2016-03-23:
Thank you. I'm sorry. I'll be more precise the next time. thank you very much.
I'm using the ev3-ev3dev-jessie-2015-12-30.img.zip image. The user and the pwd are ok, i was using the wrong one, sorry. Now i can connect with EV3 with putty. that's great!.

dlech, 2016-03-23:
Great! If you have more questions about compiling C++, just open a new issue.

closed 2016-03-23 by dlech