ev3dev/ev3dev #584: C++ compilation: how to

BorisFaz, 2016-03-24:
Hello,
I would like to start to create and compile a c++ simple program using EV3.
I want to achieve the very first "Hello world and spin a motor" program.
I succesfully downloded and  used the ev3 image ev3-ev3dev-jessie-2015-12-30.img. This debian image is working properly and i can connect to it via putty on Windows.
The communication is estamblished using a USB/Ethernet adapter and it works.
Now i don't understand what I have to do in order to create the program and run it. 
I'm reading all the pages and i'm trying to understand pages like
http://www.ev3dev.org/docs/devtools/setting-up-the-ev3dev-build-ecosystem/
and related links but for me what i want to do is compile directly on the ev3.
Any suggestion?
Thank you very much.

### System information:
    LEGO MINDSTORMS EV3
    ev3-ev3dev-jessie-2015-12-30.img

dlech, 2016-03-24:
`setting-up-the-ev3dev-build-ecosystem` is for super-advanced developers that want to work on the ev3dev operating system itself, not for people who just want to write a program.

Have you had a look at <https://github.com/ddemidov/ev3dev-lang-cpp>? It is the only C++ ev3dev library that I know of. Instructions for building on the brick are at the end of the readme.

BorisFaz, 2016-03-24:
Yes i'm using it and compiling it on the ev3 but is very slow :-( .Ok It means that https://github.com/ddemidov/ev3dev-lang-cpp is enough to start. Thanks a lot. :-)

ddemidov, 2016-03-24:
If your programs don't depend on 3rd party libraries (a hello world certainly does not), then you could use a cross-compiler. For example, [sourcery-codebench](https://www.mentor.com/embedded-software/sourcery-tools/sourcery-codebench/editions/lite-edition/) works great (it has a package for Arch linux, and probably others). You just compile your program on the main computer, copy the executable to the EV3, and it works :).

BorisFaz, 2016-03-24:
Great ddemidov. Thank you so much! :-)

closed 2017-10-24 by dlech