ev3dev/ev3dev #585: [BrickPi] tty ttyAMA0: Failed to init while setting sensors. (-110)

jabrena, 2016-03-24:
### System information:

* I am using: *Mindsensors.com PiStorms*
* I installed ev3dev using this image file: *rpi2-ev3dev-jessie-2015-12-30.img*
* My kernel version is: *4.1.15-10-rc2-ev3dev-rpi2*
* My host computer is running: *OS X*
* I am connecting using: *Wi-Fi*

### About my issue:

Doing several tests with the Board Brickpi+ with a Raspberry Pi 2, I receive the following message:

**dmesg**

Using dmesg, I couldn't see that raspi register ports from brickpi:

``` bash
[    5.665435] Registered BrickPi line discipline. (28)
[    6.611816] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.622905] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.721786] ath9k_htc 1-1.5:1.0: ath9k_htc: FW Version: 1.3
[    6.734451] ath9k_htc 1-1.5:1.0: FW RMW support: Off
[    6.734463] ath: EEPROM regdomain: 0x60
[    6.734467] ath: EEPROM indicates we should expect a direct regpair map
[    6.734478] ath: Country alpha2 being used: 00
[    6.734482] ath: Regpair used: 0x60
[    6.746193] cfg80211: Calling CRDA to update world regulatory domain
[    6.774363] ieee80211 phy0: Atheros AR9271 Rev:1
[    8.208351] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.221977] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    8.255160] 8021q: 802.1Q VLAN Support v1.8
[    9.896381] cfg80211: Calling CRDA to update world regulatory domain
[   13.056340] cfg80211: Calling CRDA to update world regulatory domain
[   15.676331] tty ttyAMA0: Failed to init while setting sensors. (-110)
[   16.216351] cfg80211: Calling CRDA to update world regulatory domain
[   19.376335] cfg80211: Calling CRDA to update world regulatory domain
[   22.536349] cfg80211: Calling CRDA to update world regulatory domain
[   25.696346] cfg80211: Calling CRDA to update world regulatory domain
[   28.856351] cfg80211: Calling CRDA to update world regulatory domain
[   32.016345] cfg80211: Calling CRDA to update world regulatory domain
[   35.176350] cfg80211: Calling CRDA to update world regulatory domain
[   38.336336] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
```

You have read all tests here:
https://github.com/ev3dev/ev3dev/issues/529

I would like to use Raspberry Pi 2 with BrickPi, but at the moment, it is impossible for me. :(

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

dlech, 2016-03-24:
Try using the official BrickPi software to make sure that this is not a hardware problem. If the official BrickPi software works, then there is something wrong with ev3dev. If the official BrickPi software does not work, then there is something wrong with your hardware.

dlech, 2016-03-24:
I saw in #529 that you said that this BrickPi works with a RPi 1 but not an RPi 2. Is this correct?

jabrena, 2016-03-24:
Hi mate,

Yes I did a test with raspi 1 with brickpi v2.8:

``` bash
[    9.382729] Registered BrickPi line discipline. (28)
[    9.395317] lego-port port0: Registered 'in1' on 'ttyAMA0'.
[    9.402480] lego-port port1: Registered 'in2' on 'ttyAMA0'.
[    9.417407] lego-port port2: Registered 'ttyAMA0:outA' on 'ttyAMA0'.
[    9.434907] lego-port port2: Added new device 'ttyAMA0:outA:lego-nxt-motor'
[    9.451017] lego-port port3: Registered 'ttyAMA0:outB' on 'ttyAMA0'.
[    9.453168] lego-port port3: Added new device 'ttyAMA0:outB:lego-nxt-motor'
[    9.474798] lego-port port4: Registered 'in3' on 'ttyAMA0'.
[    9.498394] lego-port port5: Registered 'in4' on 'ttyAMA0'.
[    9.501363] lego-port port6: Registered 'ttyAMA0:outC' on 'ttyAMA0'.
[    9.512434] lego-port port6: Added new device 'ttyAMA0:outC:lego-nxt-motor'
[    9.519286] lego-port port7: Registered 'ttyAMA0:outD' on 'ttyAMA0'.
[    9.521419] lego-port port7: Added new device 'ttyAMA0:outD:lego-nxt-motor'
[   11.395588] tacho-motor motor0: Registered 'lego-nxt-motor' on 'ttyAMA0:outA'.
[   11.436786] tacho-motor motor1: Registered 'lego-nxt-motor' on 'ttyAMA0:outB'.
[   11.459189] tacho-motor motor2: Registered 'lego-nxt-motor' on 'ttyAMA0:outC'.
[   11.481713] tacho-motor motor3: Registered 'lego-nxt-motor' on 'ttyAMA0:outD'.
```

I am doing a second test reading a sensor. 
I will send you the answer in 5m. (burning a sd card)

jabrena, 2016-03-24:
Tested with success:

``` bash
[    9.104033] Registered BrickPi line discipline. (28)
[    9.111006] lego-port port0: Registered 'in1' on 'ttyAMA0'.
[    9.113404] lego-port port1: Registered 'in2' on 'ttyAMA0'.
[    9.115720] lego-port port2: Registered 'ttyAMA0:outA' on 'ttyAMA0'.
[    9.146383] lego-port port2: Added new device 'ttyAMA0:outA:lego-nxt-motor'
[    9.160360] lego-port port3: Registered 'ttyAMA0:outB' on 'ttyAMA0'.
[    9.176318] lego-port port3: Added new device 'ttyAMA0:outB:lego-nxt-motor'
[    9.178780] lego-port port4: Registered 'in3' on 'ttyAMA0'.
[    9.178961] lego-port port5: Registered 'in4' on 'ttyAMA0'.
[    9.179135] lego-port port6: Registered 'ttyAMA0:outC' on 'ttyAMA0'.
[    9.179286] lego-port port6: Added new device 'ttyAMA0:outC:lego-nxt-motor'
[    9.179492] lego-port port7: Registered 'ttyAMA0:outD' on 'ttyAMA0'.
[    9.183896] lego-port port7: Added new device 'ttyAMA0:outD:lego-nxt-motor'
[   10.982146] tacho-motor motor0: Registered 'lego-nxt-motor' on 'ttyAMA0:outA'.
[   11.061134] tacho-motor motor1: Registered 'lego-nxt-motor' on 'ttyAMA0:outB'.
[   11.077153] tacho-motor motor2: Registered 'lego-nxt-motor' on 'ttyAMA0:outC'.
[   11.124224] tacho-motor motor3: Registered 'lego-nxt-motor' on 'ttyAMA0:outD'.
[   11.382084] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   11.387030] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   12.978623] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   12.984375] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.040749] 8021q: 802.1Q VLAN Support v1.8
[   75.188088] lego-port port0: Added new device 'in1:lego-ev3-ir'
[   75.283836] lego-sensor sensor0: Registered 'lego-ev3-ir' on 'in1'.
```

dlech, 2016-03-24:
Is this still with RPi 1?

jabrena, 2016-03-24:
I updated the test with more comments.
This test is with Raspi 1 with Brickpi v2.8

Maybe, I should update the brickpi firmware, it is another alternative, isn't it?

dlech, 2016-03-24:
I don't have a BrickPi+ to test, but I can try my BrickPi with RPi2 and see if I can reproduce the problem. I assume others have successfully used BrickPi+ with RPi2 before though. Perhaps someone can confirm?

jabrena, 2016-03-24:
Hi @DexterInd, 

do you have a document to check if I can update the firmware to last version?

Many thanks in advance

Juan Antonio

dragoncube, 2016-03-24:
@dlech I had similar problem with BrickPi+ and RPi2 (motors were operational, but not the sensors), though I didn't check dmesg, so can't say that it was precisely the same issue.

dlech, 2016-03-24:
Nope, not the same issue. FYI, you have to manually configure sensors on BrickPi using `/sys/class/lego-port`. There is not auto-detection on BrickPi.

jabrena, 2016-03-24:
Hi @dlech, 
in PiStorms occur the same, the board doesn't auto-detect the sensors. 
What is the reason?

dlech, 2016-03-25:
Because these boards were not designed to automatically detect sensors. It is not an error, it is hardware design.

dlech, 2016-05-08:
I now have this same hardware - Raspberry Pi 2 Model B and BrickPi+ (hw v2.8) - and cannot reproduce the problem. Please try with <http://www.ev3dev.org/news/2016/05/08/Kernel-Release-Cycle-11/>.

dlech, 2016-10-11:
Please re-open if this is still an issue using the latest nightly images.

dlech, 2017-01-11:
Sill a problem for @jabrena. I still cannot reproduce. Using 2016-12-21 image.

jabrena, 2017-01-14:
I tested with latest 2 images:
https://github.com/ev3dev/ev3dev/releases

```
ev3dev-jessie-2016-12-21
ev3dev-jessie-2016-10-17
``` 

For Brickpi+, the modification from the file config.txt are:

```
####################################
# Dexter Industries BrickPi config #
####################################

# uncomment both if you are using BrickPi
dtoverlay=brickpi
init_uart_clock=32000000

# uncomment only if you are using BrickPi+
dtparam=brickpi_battery=okay
``` 

In both images, I receive this message:

```
[    6.393497] Registered BrickPi line discipline. (28)
[    7.507175] smsc95xx 1-1.1:1.0 eth1: hardware isn't capable of remote wakeup
[    7.510503] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[    9.046952] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[    9.051095] smsc95xx 1-1.1:1.0 eth1: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    9.078163] 8021q: 802.1Q VLAN Support v1.8
[   16.393791] tty ttyAMA0: Failed to init while setting sensors. (-110)
``` 

I tested with 2 different BrickPi+ capes but the same hardware run on Raspberry Pi 1 & Raspberry Pi 3 without any problem:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/118
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/154

Cheers

Juan Antonio

dlech, 2017-01-14:
I am starting to think that maybe you have a bad serial port on the Raspberry Pi 2. Does the official Dexter Industries software (Rasbian for Robots) work on the combination of RPi 2 and BrickPi+? As I have already said, this combination of RPi 2 and BrickPi+ is working for me, so there must be something different about your hardware.

jabrena, 2017-01-14:
Hi @dlech, maybe yes.

The version of my Raspberry Pi 2 are:

```
Raspberry Pi 2 Model B v 1.1
Raspberry Pi 2 Model B v 1.2 <- With this one, RetroPie run but EV3Dev, Ethernet is not detected.
```

jabrena, 2017-01-14:
@dlech, 

Can you confirm that this is exactly the configuration for EV3Dev + Raspberry Pi 2?

``` 
####################################
# Dexter Industries BrickPi config #
####################################

# uncomment both if you are using BrickPi
dtoverlay=brickpi
init_uart_clock=32000000

# uncomment only if you are using BrickPi+
dtparam=brickpi_battery=okay
```

jabrena, 2017-01-14:
I am going to do this test:

```
Does the official Dexter Industries software (Rasbian for Robots) work on the combination of RPi 2 and BrickPi+? 
```

dlech, 2017-01-14:
Can you please run `cat /proc/cpuinfo | grep Revision` for these boards. I will double-check that mine is working later.

jabrena, 2017-01-14:
Of course @dlech, In 30m. I will finish both tests.

jabrena, 2017-01-14:
Hi,

Meanwhile I download 1.5GB of Raspbian for robots... I send the data from a Retropie running on both boards:

Raspberry Pi 2 Rev 1.1
```
pi@retropie:~ $ cat /proc/cpuinfo | grep Revision
Revision	: a01041
``` 

Raspberry Pi 2 Rev 1.2
```
pi@retropie:~ $ cat /proc/cpuinfo | grep Revision
Revision	: a22042
```

Later, I will send the same info with Raspbian for Robots.

jabrena, 2017-01-14:
@dlech, Take a look this table:
https://en.wikipedia.org/wiki/Raspberry_Pi

February 2015: Raspberry Pi 2, ARMv7-A (32-bit), Broadcom BCM2836, 900 MHz 32-bit quad-core ARM Cortex-A7
October 2016: Raspberry Pi 2 1.2, ARMv8-A (64/32-bit), Broadcom BCM2836, 900 MHz 64-bit quad-core ARM Cortex-A53
February 2016: Raspberry Pi 3, ARMv8-A (64/32-bit), Broadcom BCM2836, 1.2 GHz 64-bit quad-core ARM Cortex-A53

jabrena, 2017-01-15:
Info from Image from Raspbian for robots:

```
pi@dex:~ $ cat /proc/cpuinfo | grep Revision
Revision	: a01041
pi@dex:~ $ cat /proc/cpuinfo | grep Revision
Revision	: a22042
``` 

@dlech, both info is the same.

jabrena, 2017-01-15:
Finally, I repeated the tests with latest image, and the results are the same:
https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2016-12-21/ev3dev-jessie-rpi2-generic-2016-12-21.zip

```
Raspberry Pi 2 Model B v 1.1 <- Connect with SSH, but appear the classic problem:

[    7.126254] Registered BrickPi line discipline. (28)
[    8.418653] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    8.419063] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.600545] EXT4-fs (mmcblk0p2): resizing filesystem from 447488 to 957184 blocks
[    8.652244] EXT4-fs (mmcblk0p2): resized filesystem to 957184
[    9.967211] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    9.970669] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   10.015676] 8021q: 802.1Q VLAN Support v1.8
[   12.957343] random: nonblocking pool is initialized
[   17.123777] tty ttyAMA0: Failed to init while setting sensors. (-110)

Raspberry Pi 2 Model B v 1.2 <- With this one, RetroPie run but EV3Dev, Ethernet is not detected.

Impossible to connect using SSH.
```

Here another link with all revisions:
http://elinux.org/RPi_HardwareHistory#Raspberry_Pi_2_Model_B

dlech, 2017-01-26:
I just tried this with a fresh image:

```
Image file:         ev3dev-jessie-rpi2-generic-2016-12-21
Kernel version:     4.4.32-17-ev3dev-rpi2
Board:              Raspberry Pi 2 Model B Rev 1.1
Revision:           a01041
Brickman:           0.8.0
ev3devKit:          0.4.2
```

If I understand correctly, you are having problems only with `a01041`. Mine is the same and it is working, so perhaps yours is defective.

jabrena, 2017-01-26:
Mmm, it is a bit weird, because I have 2 Raspi 2 and 2 BrickPi++ so with 4 combinations, I had the same problems, but I will re test again. The same boards with the same SDCard but changing for Raspi 3 were tested without any problem, but I will re test.

Cheers

dlech, 2017-08-12:
Can this issue be closed?

jabrena, 2017-08-12:
Yes, @dlech 

any new user should use a Raspi3. :)

Cheers

closed 2017-08-12 by jabrena