ev3dev/ev3dev #588: EV3/UART sensors delay in setting mode

dlech, 2016-03-30:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: git master plus local modifications

* My host computer is running: Linux

* I am connecting using: Bluetooth

### About my issue:

#### What led up to the situation?

I'm trying to translate the official gyroboy program to ev3dev-lang-python

#### What exactly did you do (or not do) that was effective (or ineffective)?

Translated the official LEGO program verbatim to ev3dev-lang-python using the new sensor classes. 

#### What was the outcome of this action?

The robot kind of balances but eventually falls over.

#### What outcome did you expect instead?

The robot should balance.

#### Additional information:

I am using the new sensor classes, in particular `GyroSensor.rate()`. This has the effect of setting the sensor mode each time it is called. This is not a problem in of itself - it is the desired behavior. However, for EV3/UART sensors, this is a time consuming operation. It think that for EV3/UART sensors only, if we are already in the specified mode when the mode is set, we should return without reseting the mode. On other types of sensors, this behavior is desirable because it may cause a one-shot reading to update or something like that, but this is not the case for EV3/UART sensors.

dlech, 2016-03-30:
Hmm... just tried reading the value attribute only and it still has nearly the same delay. maybe this isn't a problem after all.

laurensvalk, 2016-03-31:
Remember [these two plots](https://github.com/ev3dev/ev3dev/issues/324#issuecomment-101784141) as well, causing occasional additional delay. (Fig 1: Kernel busy, Fig 2: UART sensor not sending new data.)

dlech, 2016-04-10:
Totally forgot about that. Thanks for the reminder.

closed 2017-01-14 by dlech