ev3dev/ev3dev #589: GYRO-G&A Connection timed out

mbitt, 2016-04-01:
### System information:

* I am using: *LEGO MINDSTORMS EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*

* My host computer is running: *Linux*

* I am connecting using: *USB*

### About my issue:

#### What led up to the situation?
First: Set gyro mode to GYRO-G&A.
Second: Wait a long time (i.e. some minutes)
Third: Try setting an other GYRO mode. 
Result is: Connection timed out

#### What exactly did you do (or not do) that was effective (or ineffective)?
Pluging the GYRO-SESOR out and in again seems to be the only thing that helps.
I proved all combinations of combining two modes one after the other. All other modes work like a charm. Expect the GYRO-G&A mode. See table:

```
              GYRO-ANG GYRO-RATE GYRO-FAS  GYRO-G&A GYRO-CAL
GYRO-ANG       OK!       OK!       OK!        OK!      OK!
GYRO-RATE      OK!       OK!       OK!        OK!      OK!
GYRO-FAS       OK!       OK!       OK!        OK!      OK!
GYRO-G&A     failed!   failed!   failed!      OK!     failed!
GYRO-CAL      OK!        OK!       OK!        OK!      OK! 
```
Don't know whom to blame. The driver the sensor-firmware?

Regards Martin

dlech, 2016-04-01:
I have noticed that the LEGO EV3 Gyro sensor uses lots of CPU, which I am guessing is because it sends lots of data very fast. I am not able to reproduce this with my gyro sensors, although I am using a kernel from git master, so I suppose it is possible that something has been fixed already.

I also made an interesting discovery. I have a fairly new gyro sensor and it has 2 extra modes! `TILT-RATE` and `TILT-ANG`. These seem to be the same as `GYRO-RATE` and `GYRO-ANG`, but with much less precision. Makes me wonder if LEGO put the same guts in the new WeDo 2.0 tilt sensor or something like that.

mbitt, 2016-04-07:
I tried to get access to some maybe newer GYRO-SENSORs, no success (may be next week). Meantime I wonder if there is a way to put off the current on a in-port (p.e. in3). So I don't have to plug the sensor out and in again to reset it. With (my trembling hand) this plugging gave ricochets to the sensor, which causes failures in the measured values.

closed 2016-04-07 by mbitt

dlech, 2016-04-07:
There is no need to get newer gyro sensors. I only mentioned them because of the extra modes.

My older gyro sensor works just fine, so not sure why yours would not be working too.

> I wonder if there is a way to put off the current on a in-port...

No, you cannot switch off power to a sensor. They only way is to unplug it. However, I have found that changing to another gyro sensor mode and then back to the one you want is good enough to reset the gyro sensor in most cases. Every once in a while it still needs to be unplugged, but not very often.