ev3dev/ev3dev #590: Can't set device for HiTechnic Infrared receiver on Brickpi

jfcloutier, 2016-04-02:
I can't set_device to ht-nxt-ir-receiver on the Brickpi using the latest image

`
robot@ev3dev:/sys/class/lego-port$ cd port4
robot@ev3dev:/sys/class/lego-port/port4$ cat mode
nxt-i2c
robot@ev3dev:/sys/class/lego-port/port4$ echo "ht-nxt-ir-receiver" > set_device
-bash: echo: write error: Invalid argument
`

Is the HiTechnic IR receiver supported on the BrickPi?

Thanks.

dlech, 2016-04-02:
You must also include the I2C address of the sensor. [In this case, 0x01](http://www.ev3dev.org/docs/sensors/hitechnic-nxt-irreceiver-sensor/).

    echo "ht-nxt-ir-receiver 0x01" > set_device

dlech, 2016-04-02:
If you check `dmesg | tail`, you should see something like this when you omit the address

    lego-port port0: set_device: Missing parameters

jfcloutier, 2016-04-02:
That works. Thanks again for the quick response!

closed 2016-04-02 by jfcloutier