ev3dev/ev3dev #591: Consolidate ramp_up_sp and ramp_down_sp?

dlech, 2016-04-02:
I'm working on better motor control for BrickPi. I have got to the ramping part and it doesn't work. I'm thinking that it would simplify things a bit if we only have one `ramp_time_sp` instead of separate ramp up and ramp down setpoints. Any objections?

jabrena, 2016-04-02:
Positive values should be the same effect of ramp_up_sp and negative values could be the same with ramp_down_sp, isn't it?

rhempel, 2016-04-02:
I'm not sure about removing capabilities, I can think of use cases where you want to start moving something quickly but decelerate slowly.

We have other cases where attributes change depending on the underlying driver, like rotation vs linear motors.

What about for BrickPi, you only get `ramp_sp`, but we keep the `ramp_up_sp` and `ramp_down_sp` for the LEGO motors?

I'm not sure the goal of running the exact same script on all platforms makes much sense anyways :-)

jabrena, 2016-04-02:
Hi @rhempel, I think that one of the main goals of the project is the usage of attributes in the same way for every platform. Another thing is if some platform has or not some feature, for example the Display or the usage of DC Motors in PiStorms.

rhempel, 2016-04-02:
@jabrena - well, the original goal was to create a replacement firmware for the EV3 that allowed motors, sensors, buttons, etc to be controlled using standard file read/write.

With the amazing contributions from the community, ev3dev now runs on a few different platforms, supports many LEGO compatible motor drivers, etc.

Even allowing linear and rotation motors on the EV3 introduces attributes that are available for some drivers and not others - this is normal and accepted Linux practice, we just need to be sure to document it.

I would disagree with reducing the capabilities of the core LEGO EV3 brick and motor drivers.

jabrena, 2016-04-02:
Yes, I agree with you that the project shouldn't reduce any feature from Lego EV3 Brick, so the idea from @dlech is nice but it is necessary to check if the feature reduce the usage of the motor in some scenario. In this case, I think that we should maintain the current variables, from my perspective.

dlech, 2016-04-02:
OK, forget  I said anything about BrickPi. This has nothing to do with platform limitations - I'm strictly talking KISS here. I just happen to be working with BrickPi at the moment. All of the ramping is done in the tacho motor class, so platforms/motor controllers are irrelevant for the sake of this discussion.

The point is that having only one ramp setpoint makes the implementation much more straight forward since you don't have to worry about changing speed when crossing zero. You also don't have to care about whether speed is increasing or decreasing. The same ramp algorithm works in all cases.

It also makes usage for the end user much easier. It is much easier to explain in the documentation one kind of ramping rather than a whole series of possible combinations.

Furthermore, this does not prevent you from having different ramp up and ramp down setpoints. It is as simple as

    echo 100 > ramp_time_sp
    echo 400 > speed_sp
    echo run-forever > command
    sleep 2
    echo 1000 > ramp_time_sp
    echo stop > command

dlech, 2016-04-02:
> Positive values should be the same effect of ramp_up_sp and negative values could be the same with ramp_down_sp, isn't it?

Time travel is not a feature of the EV3. We cannot have negative value for a time setpoint. :laughing:

rhempel, 2016-04-02:
Except that rampup/down is handled as part of the run to position commands - and we can't change in the middle of a position command

dlech, 2016-04-10:
OK, I guess we are keeping separate setpoints for now.

closed 2016-04-10 by dlech