ev3dev/ev3dev #592: Unable to boot evb-ev3dev-jessie-2016-01-25

void32, 2016-04-02:
### System information:

* I am using: *PASTE DEVICES HERE*

* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*

* My kernel version is: *KERNEL VERSION GOES HERE*

* My host computer is running: *OS GOES HERE*

* I am connecting using: *PASTE METHOD HERE*

### About my issue:

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

Unable to boot the latest release of the ev3 software (evb-ev3dev-jessie-2016-01-25) on my ev3 Mindstorms.
I tried to go from the latest 2015 release (ev3dev-jessie-2015-12-30) to the newest (evb-ev3dev-jessie-2016-01-25) and it turns out that the ev3 isn't able to boot this.
I have noticed that has different files in the BOOT partition.

I went back to the ev3dev-jessie-2015-12-30 and it still works find.

rhempel, 2016-04-02:
evb != ev3 :-) Double check the name of the release platform

void32, 2016-04-02:
Ok thanks. I'm sorry for bothering you.

*"I REPEAT, THIS IMAGE IS FOR BEAGLEBONE ONLY"*... I bet you got a lot of
requests for this, them :-)

rhempel, 2016-04-03:
If the issue is resolved, please close it, thks :-)

closed 2016-04-03 by void32