ev3dev/ev3dev #593: compilation of kernel on raspberry pi 1

suhrm, 2016-04-03:
Hi is there anybody who knows whether it is possible to compile the kernel directly on a raspberry pi since cross-compiling messes up some scripts  so external kernel modules can not compile

rhempel, 2016-04-03:
Yes you can cross compile on your target, but believe me, it is possible to cross compile on your host machine. The best way to do this is to set up a clean VM that you use strictly for cross compiling. I can guarantee that the rPi developers have this working - they do not compile on the rPi :-)

dlech, 2016-04-03:
from irc:

```
<suhrm> is there anybody who knows whether it is possible to compile the kernel directly on a raspberry pi since cross-compiling messes up some scripts
<dlech> suhrm, sure you can compile on Raspberry Pi. What kind of problem are you having with cross-compiling?
<suhrm> when i try to compile external modules i.e. a wifi driver i get that ./scripts/recordmcount: Syntax error: "(" unexpected
<suhrm> it seem to be a general problem when cross-compiling
<suhrm> http://stackoverflow.com/questions/31239215/scripts-recordmcount-syntax-error-unexpected-is-reported-when-i-compil
<suhrm> i have followed the guide on the github and it compiles without error, and it works without problem on the raspberry pi itself, however it seems that the kernel_source/scripts does not compile correctly if cross compiling is performed.
<dlech> it has been a long time since I have tried to cross-compile an external module. For the official ev3dev packages, I use make-kpkg to cross-compile external modules and create a debian package.
<dlech> what wi-fi driver are you trying to compile?
<suhrm> i do not get the problem when i cross-compile the module, only when i try to perform native compilation i.e. on the raspberry pi itself. I have done it before but not using a cross compiled ev3dev kernel-headers etc. The wifi dongle is a netgear a6100 and the driver is rtl8812au
<dlech> ok, I understand now.
<dlech> If you want to try compiling the kernel on the Raspberry Pi, you should be able to follow a generic guide to build a Raspberry Pi kernel and just use the ev3dev git repository instead of the official raspberry pi repository
<dlech> just don't forget to checkout the submodule with the ev3dev drivers.
<suhrm> ok i will try that
```

closed 2017-10-24 by dlech