ev3dev/ev3dev #594: [EV3] Is it impossible to set the device in EV3Brick?

jabrena, 2016-04-03:
### System information:

* I am using: *LEGO MINDSTORMS EV3*
* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*
* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*
* My host computer is running: *OS X*
* I am connecting using: *Wi-Fi*

### About my issue:
EV3 brick detect the UART sensors in automatic way but BrickPi & PiStorms do the same stuff in a manual way. My question is about this manual operation:

``` bash
echo ev3-uart > /sys/class/lego-port/port1/mode
root@ev3dev:~# echo lego-ev3-touch > /sys/class/lego-port/port1/set_device
-bash: echo: write error: Operation not supported
```
Is it an error?

I would like to know if this behaviour is an error for ev3 brick. I thought that it was possible to set the mode in every board although in lego is automatic

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

jabrena, 2016-04-03:
Related issue: https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/113

dlech, 2016-04-03:
This is not an error. The UART sensor driver on EV3 is different. It is just a generic driver and will even support unknown sensors. All information about the sensor is automatically detected and allowing set_device would never work. `set_device` is only a workaround for platforms that do not support automatic detection.

closed 2016-04-03 by dlech