ev3dev/ev3dev #595: EV3 Motor commanding, setting values delay

mob41, 2016-04-04:
### System information:

* I am using: LEGO MINDSTORMS EV3

* I installed ev3dev using this image file: ev3-ev3dev-jessie-2015-12-30.img

* My kernel version is: 3.16.7-ckt21-9-ev3dev-ev3

* My host computer is running: Windows 10

* I am connecting using: USB

### About my issue:

#### What led up to the situation?
I'm implementing a auto-reconnect function for the [ev3dev Java Unified Library](http://github.com/mob41/ev3dev-lang-java).

#### What exactly did you do (or not do) that was effective (or ineffective)?
Implemented the auto-reconnect function for all device. And tested with it. API Issue https://github.com/mob41/ev3dev-lang-java/issues/1

#### What was the outcome of this action?
My test application printed ```false``` (Disconnected) when I plugged out the motor port. But when I plugged in the motor port again, it printed ```true``` once and throws me a ```FileNotFoundException``` indicating that ```(Permission Denied)```

#### What outcome did you expect instead?
I expected that the motor won't throw a ```Permission Denied``` error and the motor will do ```run-forever``` when I plug in the port.

dlech, 2016-04-04:
First, I will ask: why would you unplug a motor or sensor while your program is running in the first place? Most user programs should not need to worry about this. My opinion is that if something comes unplugged while your program is running, the program should crash. Then you fix the problem and start the program again.

If you really want to have hotplug capabilities in your library, have a look at binding [libudev](https://www.kernel.org/pub/linux/utils/kernel/hotplug/libudev/ch01.html) (if there are not already java bindings for it). You can look at the [ev3devKit source code](https://github.com/ev3dev/ev3devKit/tree/ev3dev-jessie/src/devices) to see an example of how libudev is used.

When a device is reconnected, it is a new device in sysfs, so you need to create a new class instance for it. You cannot reuse the old instance.

jabrena, 2016-04-04:
@mob41 Hi, I suppose that the scenario that you say is about Sensor/Motor testing. For robots that scenario that you say is not necessary but for Sensor Testing, you could create some kind of utility.

Anyway, you are using a old version of the library. (v0.1.0)
Take a look current syntax in this example:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/126

Can you update the project? Create a issue in the project and I will help you.
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues

In 1/2 weeks, I will release a new version with support for PiBoards.

Cheers

jabrena, 2016-04-04:
Sorry @mob41 I observed that you are implementing your own Java library, nice!

mob41, 2016-04-04:
@dlech So how the ```connected``` variable works at all? My code was "pinging" though all the motors/sensors to see the equal addresses to determine is the device connected.

However, I am quite busy today as I have my school and studying. I might have you a late reply.

dlech, 2016-04-04:
I don't know. The `connected` variable is an `ev3dev-lang` thing, not a general ev3dev thing. Better ask at <https://github.com/ev3dev/ev3dev-lang/issues>.

WasabiFan, 2016-04-05:
"connected" just signifies if the device believes it's ready to communicate. That means that it has found a device that is connected to the requested port. If reading or writing a property fails, `connected` should be changed to false. See [here](http://ev3dev-lang.readthedocs.org/en/latest/classes.html#device-abstract).

mob41, 2016-04-05:
Hello.

Okay I will change the ```connected``` variable later on. However from my issue, I couldn't set/get property from the motor <b>immediately</b> from the system class, as shown above, it said Permission Denied.

ddemidov, 2016-04-05:
> it said Permission Denied.

That is probably because you did not recreate your device class. Say, when you have single motor in the ev3, it has an entry in sysfs like this:
```
$ ls /sys/class/tacho-motor/
motor0
```

When you disconnect the motor and connect it again, its index (0) changes to the next one:
```
$ ls /sys/class/tacho-motor/
motor1
```

And your class still tries to work with `/sys/class/tacho-motor/motor0` which no longer exists.

mob41, 2016-04-05:
Nope, the program tried read/write with another class. Linux will say ```No such file exist``` if the class no longer exist

mob41, 2016-04-05:
You can try compiling on the commit https://github.com/mob41/ev3dev-lang-java/commit/42d96764f4cdedcf531ba6077551c1978aeb0c81 And the main class is ```ev3dev.tests.TestOnEV3``` and run the jar on your EV3. I am putting that jar now.

ddemidov, 2016-04-05:
In this case, can you try and follow the [tacho motor tutorial](http://www.ev3dev.org/docs/tutorials/tacho-motors/) from the command line? Will that work for you after he motor is reconnected?

mob41, 2016-04-05:
Yup it works.

ddemidov, 2016-04-05:
It seems then the problem is with your implementation of the library. Do you use file caching? That could contain stale file handles.

mob41, 2016-04-05:
As I said above, the program can find the new motor with a "specified" delay.

mob41, 2016-04-05:
:weary: However, this function is no-longer required. But this can be a kernel discussion about why it throwing a permission denied.

ddemidov, 2016-04-05:
If the command line tutorial works as you say, then this is not a kernel problem at all.

dlech, 2016-04-05:
> I couldn't set/get property from the motor immediately from the system class, as shown above, it said Permission Denied.

this is expected. udev (in userspace) is used to change the group id so there is some indeterministic  delay after the device appears before the permissions are set correctly. This is why I suggested using libudev. It will notify you that a device was connected, hopefully after the gid has been changed.

closed 2016-04-06 by mob41