ev3dev/ev3dev #596: Hang during boot

pambrose, 2016-04-05:
### System information:

* I am using: BrickPi+ with a Raspberry Pi1 model B

* I installed ev3dev using this image file:rpi-ev3dev-jessie-2015-12-30.img

* My kernel version is: unknown at this point

* My host computer is running: OSX

* I am connecting using: WiFi

### About my issue:

When booting the downloaded image, it hangs after the statement:
[  OK  ] Started Update UTMP about System Runlevel Changes

I tried using the VGA settings for the HDMI output incase it was crashing in the video driver.

I rebooted numerous times and it consistently hangs.

I am unable to reach it via ssh because it has not started the ssh daemon yet.

Any thoughts would be appreciated.

Paul

dlech, 2016-04-05:
I've noticed a hang on boot if I have not enabled the brickpi overlay in `config.txt` (and have pistorms enabled instead). It boots fine if the brickpi overlay is enabled though.

dlech, 2016-04-14:
I have made some changes that I think will fix this. The issue I fixed is that if communications failed in the battery driver (from selecting incorrect settings in config.txt or some other reason), the system would get locked up because it kept trying to read the battery voltage in an endless loop. The fix has been released in kernel `v4.1.21-10-ev3dev-rpi` but has not been released in an SD card image yet.

pambrose, 2016-04-18:
Weird.  I rewrote the SD card image and tried every permutation of the BrickPi values in config.txt.  I hang in the same place every time.

dlech, 2016-04-18:
Did those permutation include leaving all of the BrickPi options disabled? Also try booting without the BrickPi or any USB devices connected.

pambrose, 2016-04-18:
I left all the options disabled and encountered the same hang.  I removed the BrickPi and the wifi dongle and left all options disabled and it still hangs!  Strange. Have you tested with a Model B+ V 1.2 board?

dlech, 2016-04-18:
No, I don't have a B+, just a B and a 2B (and soon will have 0 and 3B).

Maybe try a different SD card. And a different RPi if you have access to one.

pambrose, 2016-04-18:
Ordering a different RPi now.  Thanks.

pambrose, 2016-04-20:
I tried a RPi 2 and saw the same behavior.  But it turns out that sshd was running on the RPi, so I am able to reach the device.  You might want to add a note somewhere warning people to not wait for the login prompt on the console.  Thanks for what you are doing with this project.  It is awesome.

closed 2016-04-20 by pambrose