ev3dev/ev3dev #597: Can not recognize wifi modal

Tom-Shen, 2016-04-06:
### System information:

* I am using: *LEGO MINDSTORM EV3*

* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*

* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*

* My host computer is running: *OS X 10.11*

* I am connecting using: *Bluetooth*

### About my issue:

#### What led up to the situation?
I got a CSR Chipset wifi modal, it can work under the default EV3 system. But when I boot into ev3dev, it does not recognize the modal.

#### What exactly did you do (or not do) that was effective (or ineffective)?
I used ```sudo connmanctl``` and then use enable wifi. it does not work and have this error:
```Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist```

#### What was the outcome of this action?
The error stated above.

#### What outcome did you expect instead?
I expected the wifi to be enabled.

dlech, 2016-04-07:
What is the output of `lsusb` on the EV3? I'm guessing we just need to figure out which driver this uses and enable it.

Tom-Shen, 2016-04-07:
Here is the output of ```lsusb```:
Bus 001 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

dlech, 2016-04-07:
This says it is a Bluetooth dongle, not Wi-Fi.

closed 2016-10-11 by dlech