ev3dev/ev3dev #598: "Access Denied" issue

Johnny-West, 2016-04-09:
I'm having a problem with my EV3dev thing. It keeps saying "Access Denied" on PuTTY. Could you fix it?
[image]

dlech, 2016-04-09:
Please answer the questions from the issue template: 

### System information:

* I am using: *PASTE DEVICES HERE*

* I installed ev3dev using this image file: *IMAGE NAME GOES HERE*

* My kernel version is: *KERNEL VERSION GOES HERE*

* My host computer is running: *OS GOES HERE*

* I am connecting using: *PASTE METHOD HERE*

### About my issue:

#### What led up to the situation?

#### What exactly did you do (or not do) that was effective (or ineffective)?

#### What was the outcome of this action?

#### What outcome did you expect instead?

Johnny-West, 2016-04-09:
System information:

I am using: LEGO Mindstorms EV3
I installed ev3dev using this image file:  ev3dev-jessie-2015-05-01
My kernel version is: 3.16.7-ckt10-4-ev3dev-ev3
My host computer is running: Windows 10
I am connecting using: SSH

About my issue:

What led up to the situation? Well, I was trying to get my EV3 up and running using SSH (as explained in the Getting Started" part of the EV3dev page.

What exactly did you do (or not do) that was effective (or ineffective)? As I said, tried using SSH

What was the outcome of this action? Ended up in "Access Denied".

What outcome did you expect instead? Fix the password to "maker".

dlech, 2016-04-09:
This is an old image file. Please try with the most recent. Currently [ev3dev-jessie-2015-12-30](https://github.com/ev3dev/ev3dev/releases/tag/ev3dev-jessie-2015-12-30).

WasabiFan, 2016-04-10:
For reference, I believe that the 2015-12-30 release was [the first](http://www.ev3dev.org/news/2015/12/30/ev3dev-jessie-2015-12-30-release/) to use the new `robot:maker` login info. Images from before that used a different user name and password. So, because you're using an old release, the new credentials don't work. Re-imaging using the newer image as @dlech suggested should get you the new account.

closed 2016-07-24 by dlech