ev3dev/ev3dev #599: Python API install problem

Johnny-West, 2016-04-10:
I'm having trouble setting up my Python API. Could you help me?
[image]

dlech, 2016-04-10:
Not exactly sure what you are trying to do. This is why we ask you to answer all of the questions each time you open a new issue.

ev3dev-lang-python comes preinstalled so there is nothing to setup. If you have further questions about ev3dev-lang-python, have a look at https://github.com/rhempel/ev3dev-lang-python and open an issue there if you get stuck.

Johnny-West, 2016-04-10:
I'm trying to get a python script into the EV3.

WasabiFan, 2016-04-10:
@Johnny-West Although it would be great if you could provide some more information, I think I might be able to help you. Try running `sudo apt-get install git` and then try what you were doing again.

Johnny-West, 2016-04-10:
It isn't working. Any other solutions?

sudharsh, 2016-04-14:
I also noticed that you are doing an `import` from the shell, hence the command not found error. Try running `python` from the shell which will drop you in the interpreter. Your import statements will work then.

closed 2017-10-24 by dlech