ev3dev/ev3dev #600: Remove encoder_polarity attribute in tacho-motor class

dlech, 2016-04-10:
As it currently stands, the encoder polarity is selected by selecting a motor driver (e.g. `fi-l12-ev3-50`). There is really no longer a need for the `encoder_polarity` attribute to be exposed to userspace.

rhempel, 2016-04-10:
I agree - the encoder polarity was helpful when we did not have specific motor drivers. It's no longer needed so I'm OK with getting rid of it.

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (an other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech