ev3dev/ev3dev #601: tacho motor polarity attribute is broken

dlech, 2016-04-10:
With all of the recent tacho-motor class changes, the `polarity` attribute is quite broken everywhere. In the past, we passed the polarity to the low level driver, but this is no longer the case.

I'm thinking that to keep things simple, we just make the polarity attribute change the sign of any attribute that has to do with position or speed. For example, if the motor position is `1234` and we invert the polarity, reading the `position` attribute will return `-1234`. However, internally, the position will still be stored as `1234` so that it does not mess up the motor control if the motor is currently running. In other words, the polarity attribute will be implemented completely in the attribute store and show functions in the tacho motor class like this:

    val = tm->ops->get_value(tm->context);
    if (tm->params.polarity == DC_MOTOR_POLARITY_INVERSED)
        val *= -1;
    return sprintf(buf, "%d\n", val);

Related: #545

rhempel, 2016-04-10:
Sounds right!

dlech, 2016-04-11:
This fix has been released in the `3.16.7-ckt26-10-ev3dev-ev3` kernel (an other `10-ev3dev` kernels on other platforms). Be aware the tacho-motor class has breaking changes in this release, so don't expect motors to work with your existing programs/libraries if you upgrade to this version.

closed 2016-04-11 by dlech