ev3dev/ev3dev #602: Linux kernel/Git downloadissues

Johnny-West, 2016-04-10:
I'm having problems with my EV3 build because it contains some errors inside the Linux kernel. Can you fix this?
[image]

dlech, 2016-04-10:
Make sure your EV3 is connected to the Internet, then run

    sudo apt-get update

then run `sudo apt-get install` again.

Johnny-West, 2016-04-10:
I tried, and they said:

```
Err http://ev3dev.org jessie InRelease

Err http://ftp.debian.org jessie Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://ev3dev.org jessie Release.gpg
  Temporary failure resolving 'ev3dev.org'
Reading package lists... Done
W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/InRelease

W: Failed to fetch http://ftp.debian.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ftp.debian.org'

W: Failed to fetch http://ev3dev.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'ev3dev.org'

W: Some index files failed to download. They have been ignored, or old ones used instead.

```

So this is a big problem. Would you mind fixing it?

rhempel, 2016-04-10:
First, you need to make sure that your EV3 is actually able to access the Internet - can you `ping` anywhere outside your local network?
Have you followed the instructions in http://www.ev3dev.org/docs/tutorials/ related to connecting to the Internet under Networking?

psychemedia, 2016-04-16:
I'm having terrible problems with updating packages - networking is fine (I can install some things) but the packages I really need are just coming back as `E: Unable to locate package WHATEVER`

The `apt-get` update also fails to complete cleanly... It hangs at the end:

```
Ign http://ftp.debian.org jessie InRelease         
Hit http://ftp.debian.org jessie Release.gpg       
Hit http://ftp.debian.org jessie Release
Hit http://ev3dev.org jessie InRelease                 
Get:1 http://ftp.debian.org jessie/main armel Packages [6,622 kB]
Hit http://ev3dev.org jessie/main armel Packages        
Ign http://ev3dev.org jessie/main Translation-en_US                            
Ign http://ev3dev.org jessie/main Translation-en                               
Get:2 http://ftp.debian.org jessie/contrib armel Packages [37.1 kB]            
Get:3 http://ftp.debian.org jessie/non-free armel Packages [59.6 kB]           
Hit http://ftp.debian.org jessie/contrib Translation-en                        
Get:4 http://ftp.debian.org jessie/main Translation-en [4,581 kB]              
Get:5 http://ftp.debian.org jessie/non-free Translation-en [72.5 kB]           
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]                          
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]^Crobot@ev3dev:~$
```

It would be really handy if the distribution included `git`, `pip` and `python-dev` (maybe `build-essential ` as well?)

dlech, 2016-04-16:
`sudo apt-get update` should fix you up.

We will be including `git` by default in the next image release. I believe `easy_install` is already included. Building on the EV3 is so slow, we don't bother including `build-essential` or `python-dev`. Feel free to open a new issue if you think they should be included.

psychemedia, 2016-04-16:
I've run `sudo apt-get update` repeatedly and it doesn't help.... `git` is one of the packages not recognised, as is `python-dev`. I had things working with an old version of ev3dev this morning, but since upgrading to a reflashed `ev3dev` at lunch time I've just gone backwards through not being able to load packages. (It took 90 mins of dancing between `easy_install` and `pip` (the latter loaded from source) to get `ipython` installed, and the kernels still don't run properly because I can't build `pyzmq`.)

dlech, 2016-04-16:
Have you waited for `apt-get update` to finish without aborting it? It does take a very long time - especially the first time you run it, you just have to wait and it will eventually finish.

Debian has an [older version of IPython](https://packages.debian.org/hu/source/jessie/ipython) already packaged. Would this work?

psychemedia, 2016-04-16:
I ended up leaving it to update overnight and it seems to have run to completion at some point during that period  - thanks... I guess I was too hasty assuming it was stuck when it had apparently stalled for >5000s at a time. Doh!

moriarty, 2016-04-16:
At the beginning I let installations and compiles run overnight on the brick... But watch out, sometimes they'll run out of ram & swap.

It is usually quicker to set up a brickstrap environment, sync your /home/robot/ directories (or anywhere else you were working), install or compile what you need and create a new SD card, or copy what you need form the brickstrap environment back to the brick.

psychemedia, 2016-04-16:
@moriarty Thanks for the brickstrap tip - I'll explore that. Is there a base dockerfile anywhere that can kickstart that process?

moriarty, 2016-04-16:
Not to my knowledge. Actually I was thinking of making a docker image earlier today- but then got distracted with another docker project I stumbled on

online4ever, 2016-04-26:
robot@ev3dev:~$ sudo apt-get update
[sudo] password for robot:
Get:1 http://ftp.debian.org jessie InRelease
100% [1 InRelease gpgv 5,620 B]Splitting up /var/lib/apt/lists/partial/ftp.debiaIgn http://ftp.debian.org jessie InReleaseata and signature failed
E: GPG error: http://ftp.debian.org jessie InRelease: Clearsigned file isn't valid, got 'NODATA' (does the network require authentication?)
robot@ev3dev:~$ ping google.de
PING google.de (172.217.18.163) 56(84) bytes of data.
From p4FE3A785.dip0.t-ipconnect.de (79.227.167.133) icmp_seq=1 Packet filtered

... does anybody know what is going wrong?

User is robot and password is maker?

dlech, 2016-04-26:
Maybe try a different debian mirror<sup>[1]</sup> or use the HTTP redirector<sup>[2]</sup>. Edit `/etc/apt/sources.list` as root (e.g. `sudo nano /etc/apt/sources.list` and change `ftp.debian.org` to `httpredir.debian.org`.

[1]: https://www.debian.org/mirror/list
[2]: http://http.debian.net/

online4ever, 2016-04-26:
Even I changed ... same result.

dlech, 2016-04-26:
Maybe your ISP is blocking the file. Can you browse to <http://archive.ev3dev.org/debian/dists/jessie/InRelease>? (or `wget http://archive.ev3dev.org/debian/dists/jessie/InRelease` on the EV3)

rhempel, 2016-04-26:
If the ISP is blocking the file, then it those commands might fail on his main computer as well, or there could be a firewall on his shared internet connection preventing the download.

moriarty, 2016-04-26:
You're in Germany?

I know that some university networks block port ranges on the Eduroam networks (including ssh + ftp, which is really annoying) and that some student dormitory networks block shared internet connections (studentenwerk Bonn)

online4ever, 2016-04-27:
Hi all,

yes I come from Germany a village near Karlsruhe called Leopoldshafen. I have an ordinay T-Online DSL 16 MBit access over a Fritz!Box 7490. So nothing special at all. Hear are my wget results ...

robot@ev3dev:~$ wget http://archive.ev3dev.org/debian/dists/jessie/InRelease
--2016-04-27 14:58:56--  http://archive.ev3dev.org/debian/dists/jessie/InRelease
Resolving archive.ev3dev.org (archive.ev3dev.org)... 206.190.152.46
Connecting to archive.ev3dev.org (archive.ev3dev.org)|206.190.152.46|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://fritz.box/tools/kids_not_allowed.lua?account=ip-192.168.178.39 [following]
--2016-04-27 14:58:56--  http://fritz.box/tools/kids_not_allowed.lua?account=ip-192.168.178.39
Resolving fritz.box (fritz.box)... 192.168.178.1
Connecting to fritz.box (fritz.box)|192.168.178.1|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘InRelease’

InRelease               [ <=>                  ]   5.49K  --.-KB/s   in 0.003s

2016-04-27 14:58:57 (2.07 MB/s) - ‘InRelease’ saved [5620]

robot@ev3dev:~$ sudo apt-get update
[sudo] password for robot:
Get:1 http://httpredir.debian.org jessie InRelease
100% [1 InRelease gpgv 5,620 B]Splitting up /var/lib/apt/lists/partial/httpredirIgn http://httpredir.debian.org jessie InReleaseata and signature failed
E: GPG error: http://httpredir.debian.org jessie InRelease: Clearsigned file isn't valid, got 'NODATA' (does the network require authentication?)
robot@ev3dev:~$

moriarty, 2016-04-27:
The:
> fritz.box/tools/kids_not_allowed.lua

makes me think you're fritz box has been configured with parental settings blocking access.

markyi370, 2016-05-08:
@online4ever Maybe you can explain your situation and have the admin open it up for you.

online4ever, 2016-05-09:
Yes. Good idea. At the moment everything works fine. Thanks a lot ;-)

psychemedia, 2016-05-16:
@moriarty If there isn't a Dockerfile, do you know if there is a Vagrantfile that sets up a machine for building images under brickstrap?

closed 2017-10-24 by dlech