ev3dev/ev3dev #605: Support for BrickPi with Raspberry Pi 3

JorgePe, 2016-04-11:
After closing #563 tried BrickPi with Raspberry Pi3. As expected, it doesn't work.

But...
- I downloaded latest firmware from Raspbian (https://github.com/raspberrypi/firmware/tree/master/boot)
- replaced these files in /boot/flash:
bcm2709-rpi-2-b.dtb
bootcode.bin
fixup.dat
fixup_cd.dat
fixup_db.dat
fiuxp_x.dat
start.elf
start_cd.elf
start_db.elf
start_x.elf
- also copied this new file: bcm27610-rpi-3-b.dtb
- also copied all content of 'overlays' folder
- added this line to /boot/flash/config.txt 
dtoverlay=pi3-miniuart-bt-overlay

after reboot I have /dev/sys/tacho-motor/motor0 to 4 mapped to ttyAMA9:MA to MD devices

Strange, I can read EV3 medium motor position but cannot make it rotate (it works with Pi2). I'm powering BrickPi from the Pi3, will try with external power later.

dlech, 2016-04-11:
> I downloaded latest firmware from Raspbian

The `rpi-firmware` package in ev3dev was updated to this same firmware yesterday, so should be no difference here. Likewise, the .dtb files (including overlays) are in the current ev3dev rpi kernel package and installed by `flash-kernel` so should be no difference there.

> added this line to /boot/flash/config.txt dtoverlay=pi3-miniuart-bt-overlay

This should work without having to copy all of the things you did. Can you test this?

dlech, 2016-04-11:
> I'm powering BrickPi from the Pi3, will try with external power later.

This is most likely the problem.

dlech, 2016-04-11:
> added this line to /boot/flash/config.txt dtoverlay=pi3-miniuart-bt-overlay

Need to add this line to https://github.com/ev3dev/brickstrap/blob/master/projects/ev3dev-jessie/rpi-base/root/boot/flash/config.txt

@JorgePe, can you post the default config.txt that ships with the RPi3?

dlech, 2016-04-11:
More on tty renaming: https://github.com/raspberrypi/linux/issues/1399

JorgePe, 2016-04-11:
Uff, you're fast. The line that worked for me is

`dtoverlay=dtoverlay=pi3-disable-bt`

(I had changed in the original post but you replied before I complete it)

I replaced all files from /boot/flash from backup, but left all overlay folder files. It doesn't work, no /sys/class/tacho-motor folder.

But if I add 'bcm27610-rpi-3-b.dtb' again (and just this file) it works again.

I'm downloading last raspbian image to be sure about the default config.txt file

dlech, 2016-04-11:
> But if I add 'bcm27610-rpi-3-b.dtb' again (and just this file) it works again.

This file should be automatically added if you run `sudo flash-kernel` on the RPi3. If not, there is a bug in the `flash-kernel` package (so open a new issue if it does not work).

JorgePe, 2016-04-11:
'/boot/config.txt' from Raspbian (2016-03-18)
```
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
```

JorgePe, 2016-04-11:
> >    I'm powering BrickPi from the Pi3, will try with external power later.
> This is most likely the problem.

Nop. Using just external power or using both (USB and external power) is the same, I can read motor0 but not make it turn.

```
echo 100 > duty_cycle_sp
echo run-forever > command
```

Tried another medium EV3 motor, same behavior. Need to find other my other EV3 devices to test.

dlech, 2016-04-11:
The motor drivers have been changed. `run-forever` no longer looks at `duty_cycle_sp`. 

    echo 100 > duty_cycle_sp
    echo run-direct > command

or 

    echo 1000 > speed_sp
    echo run-forever > command

should work.

JorgePe, 2016-04-11:
Oops... missed that.
Was following [http://www.ev3dev.org/docs/tutorials/tacho-motors/] it was worked with RPi2 because I only upgraded today. Things changing to fast for me to follow up :)

Thanks any way, this day was very productive.

closed 2016-04-11 by JorgePe