ev3dev/ev3dev #606: flash-kernel: unsupported platform in Raspberry Pi3

JorgePe, 2016-04-11:
### System information:

* I am using: Rapsberry Pi Model 3B, Dexter Industries BrickPi

* I installed ev3dev using this image file: rpi2-ev3dev-jessie-2015-12-30.img.xz

* My kernel version is: 4.1.21-10-ev3dev-rpi2

* My host computer is running: Ubuntu Linux

* I am connecting using: Wi-Fi

### About my issue:

#### What led up to the situation?
I'm testing Raspberry Pi 3 with BrickPi, and issue #605 sent me here
I deleted 'bcm2710-rpi-3-b.dtb' from /boot/flash/ and wanted to try command 'flash-kernel'

#### What exactly did you do (or not do) that was effective (or ineffective)?
I ran
`sudo flash-kernel`

#### What was the outcome of this action?
`Unsupported platform.`

#### What outcome did you expect instead?
flash-kernel should re-create 'bcm2710-rpi-3-b.dtb'

dlech, 2016-04-11:
What is output of `dpkg -s flash-kernel`?

dlech, 2016-04-11:
Just the version actually. `dpkg -s flash-kernel | grep Version`

JorgePe, 2016-04-11:
`Version: 3.35ev3dev8`

dlech, 2016-04-11:
hmm... did output of `cat /proc/device-tree/model; echo` change? It should match `Machine:` in https://github.com/ev3dev/flash-kernel/commit/e11ba4de6f82b459b3996dc19729bb84577ea664

dlech, 2016-04-11:
Oops, wrong link: Correct link is https://github.com/ev3dev/flash-kernel/commit/ddfcc0c3dc5544ddce058f6dc8c9f31e37c01184

JorgePe, 2016-04-11:
```
cat /proc/device-tree/model; echo
Raspberry Pi 3 Model B Rev 1.2
```
seems ok

dlech, 2016-04-11:
I wonder if I accidentally copied some non-printing character or something. You could try editing `/usr/share/flash-kernel/db/all.db` and delete that line and re-create it.

JorgePe, 2016-04-11:
```
robot@ev3dev:~$ sudo flash-kernel 
Installing bcm2710-rpi-3-b.dtb into /boot/dtb-4.1.21-10-ev3dev-rpi2
Taking backup of dtb-4.1.21-10-ev3dev-rpi2.
Installing new dtb-4.1.21-10-ev3dev-rpi2.
flash-kernel: installing version 4.1.21-10-ev3dev-rpi2
Version: Linux version 4.1.21-10-ev3dev-rpi2 (david@freyr) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1 SMP Sun Apr 10 21:33:51 CDT 2016
DT: y
283x: n
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel7.img.
Installing new kernel7.img.
Taking backup of bcm2710-rpi-3-b.dtb.
Installing new bcm2710-rpi-3-b.dtb.
```

Nice! Thanks a lot, case closed.

dlech, 2016-04-11:
Did you change something to fix it?

JorgePe, 2016-04-11:
No.
Just made what you said: deleted everything from 'Machine:' to end of line, then past the output I posted above, then deleted the end of the line and inserted a new line just in case.

dlech, 2016-04-11:
OK, so you did change something :wink:

So we shouldn't close this issue until the `flash-kernel` package is fixed.

dlech, 2016-04-11:
I don't understand what you changed. If I try to delete that line and re-create it, git does not pick up the diff.

JorgePe, 2016-04-11:
I cannot see any difference either.
Could it be just the "touch" in the file? Perhaps I had something wrong with the file, not with the content.

dlech, 2016-04-11:
not sure. you could try reinstalling and see if it breaks again `sudo apt-get install --reinstall flash-kernel`

JorgePe, 2016-04-11:
I reinstalled. Still works.

dlech, 2016-04-11:
OK. I guess we will close unless someone can reproduce the problem again.

closed 2016-04-11 by dlech