ev3dev/ev3dev #607: Built-in Bluetooth on RPi 3

dlech, 2016-04-11:
Here is what I know so far about what it will take to get built-in blueooth working on RPi3

- [ ] We need a udev rule like [this](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/etc/udev/rules.d/99-com.rules) to create an alias for the Bluetooth UART
- [ ] We need a systemd service like [this](https://github.com/ev3dev/ev3-systemd/blob/ev3dev-jessie/systemd/ev3-bluetooth.service) to run hciattach for us.
- [ ] Need to add `dtoverlay=pi3-miniuart-bt` to `config.txt` if using BrickPi

Would also be useful to see what `hciuart.service` from the official RPi distro looks like.

cc: @JorgePe

BertLindeman, 2016-04-11:
From my Rpi3 
```
uname -a
Linux rpi3 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
```

'cat /lib/systemd/system/hciuart.service'
```
[Unit]
Description=Configure Bluetooth Modems connected by UART
ConditionPathIsDirectory=/proc/device-tree/soc/gpio@7e200000/bt_pins
Before=bluetooth.service
After=dev-ttyAMA0.device

[Service]
Type=forking
ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

[Install]
WantedBy=multi-user.target

```

Was this what you wanted?

dlech, 2016-04-11:
Yes, thank you!

dlech, 2016-06-26:
Looks like this will be a bit more complicated. We have to patch bluez as well.

https://github.com/raspberrypi/linux/issues/1314

dlech, 2016-06-28:
More useful info: http://www.slideshare.net/yeokm1/raspberry-pi-3-uartbluetooth-issues

dlech, 2016-06-28:
Just about there. Will have to wait for the next kernel release for the last piece of the puzzle though.

Other pieces are the new `rpi-systemd` package (replaces `brickpi-systemd`) and the config.txt changes from the brickstrap commit above. Oh, and the firmware in the `firmware-rpi` package and the patched `bluez` package. Since the current debian version of the `bluez` package has `+b1` because it was rebuilt for some reason, you actually have to "downgrade". `sudo apt-get install bluez=5.23-2ev3dev1`

dlech, 2016-07-24:
Fix has been released in the `13-ev3dev` kernels.

dlech, 2016-07-24:
@jorgepe @BertLindeman 

Please try the latest nightly image build to confirm that built-in bluetooth is working out of the box.

 [ [image] ](https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion)

BertLindeman, 2016-07-25:
Downloaded from `https://bintray.com/ev3dev/nightly/ev3dev-jessie/_latestVersion`

setup the image ```rpi2-nightly-2016-07-24-bootroot.img```
and network and such.

```
uname -a
Linux ev3dev 4.4.15-13-ev3dev-rpi2 #1 SMP Sun Jul 24 17:08:21 CDT 2016 armv7l GNU/Linux
```

On windows remove the previously known ev3dev bluetooth device before trying to connect. 

Then in `connmanctl` :
```
agent on
enable bluetooth
```

as root: `hciconfig hci0 up`

use `bluetoothctl` to pair 
```
agent on
default-agent
scan on
discoverable on
pairable on
pair xx:xx:xx:xx:xx:xx
```

bluetoothctl Request confirmation =-> yes

connect in connmanctl
 
It took me more time than shows here, 
as I used to use bt-network to setup the device
but bottomline: it works.

dlech, 2016-07-25:
Great! Thanks for testing!

closed 2016-07-25 by dlech

JorgePe, 2016-07-27:
@dlech a little late but might have some interest: both internal BT and internal wifi work out of the box, I no longer need the Wi-Fi USB dongle.
After enabling BT with "connmanctl enable bluetooth" I have a 4.1 BT device.

But when enabling BrickPi in config.txt, BT no longer works - no such service in connmanctl. But wi-fi still works.

JorgePe, 2016-07-27:
Sorry, should have read the config.txt better.
BrickPi does work with Raspberry Pi 3 onboard BT and onboard Wi-Fi, just needed to also uncomment these 2 lines:
`dtoverlay=pi3-miniuart-bt`
`core_freq=250`

dlech, 2016-07-27:
Awesome! Thanks for testing. If there is something we can do to make the `config.txt` instructions more clear, open a new issue with your suggestions.

JorgePe, 2016-07-27:
I think the instructions are fine, at least for the kind of people that use ev3dev with a BrickPi on a Raspberry Pi3 :)
I knew that the onboard BT uses by default the serial port and probably would need to use the soft alternative. But somehow forgot that.

Later I will try to test it better with my BLE python scripts, I'm not sure if miniuart afects BT/BLE timmings, hope not.