ev3dev/ev3dev #608: update ev3dev-lang-python for kernel changes

dlech, 2016-04-11:
Since `ev3dev-lang-python` ships pre-installed, we need to get it updated before we can do an image release.

rhempel, 2016-04-12:
@dlech - I'm travelling on business for the next two weeks - I have the EV3 with me and I want to run some experiments with local attributes before I release a new ev3dev-lang-python

laurensvalk, 2016-04-17:
> Note that currently, the source is only installed in the default Python 2.x location - this will be addressed in the next package we release.

Will it also be pre-installed for Python 3?

dlech, 2016-04-17:
yes

psychemedia, 2016-05-08:
@dlech I appreciate you and any other maintainers have other calls on your time but is there a timeline on when py bindings may work with latest ev3dev master? I need to commit to a version for in order to produce course materials for an HE course in next 10 days or so that will then gold master in 4 weeks or so, so if it's unlikely to be in that period I know I should roll back and can start on it now.

Thanks - and all your efforts to date and in the future much appreciated..:-)

dlech, 2016-05-08:
Sorry, but we can't commit to any time frame. I would recommend to play it safe an use the `ev3dev-jessie-2015-12-30` image for your course (and don't upgrade any packages). Even if we were to get the new version released before then, it has not been widely tested and could be buggy.

rhempel, 2016-05-10:
@psychemedia  if you're willing to do help with some testing, then I can work on getting a compatible binding out. I just built and installed the ckt10 kernel on Saturday, and updated the `spec.json` file that helps generate bindings. I can maybe get branch of `ev3dev-lang-python` that's compatible with ckt11 later tonight or tomorrow...

dlech, 2016-05-10:
I think you mean `3.16.7-ckt26-10-ev3dev-ev3`. There is no such thing as a `ckt10` kernel. But, you should be using <www.ev3dev.org/news/2016/05/08/Kernel-Release-Cycle-11/>.

rhempel, 2016-05-10:
Yes, you are correct - I have all evening today to get the next Python binding ready

psychemedia, 2016-05-11:
@rhempel I just rolled back to the December kernel but will get a new SD card run in parallel and try out the new bindings as my course activities come together.

ValorNaram, 2016-05-16:
I have programmed a python library for the LEGO MINDSTORMS EV3, but I have a problem!
I cannot access /sys/class/tacho-motor/motor0/command with **sudo** privilegs.
My script is based on http://www.ev3dev.org/docs/drivers/tacho-motor-class
But it doesn't works, because I can't access the command file with **sudo** privilegs.

WasabiFan, 2016-05-16:
@ValorNaram please open a new issue to ask your question, as it seems unrelated to this thread.

dlech, 2016-07-26:
New package is available for testing. It's python 3 only, so it is a new package

    sudo apt-get install python3-ev3dev

rhempel, 2016-07-26:
@dlech - how do we go about ensuring that the existing python-ev3dev package is marked as deprecated?

dlech, 2016-07-26:
There isn't a way to do this. We can't change an existing package. However, starting with 2016-07-26 nightly image, the `python-ev3dev` package is omitted and the `python3-ev3dev` package is included.

We just need to update the documentation everywhere to use python 3.

rhempel, 2016-07-26:
We could build a package that installs a Python script that says upgrade to Python 3 :-)

dlech, 2016-07-26:
doesn't seem like it is worth the effort.

rhempel, 2016-07-26:
You're probably right - we'll just make all the docs refer to `python3-ev3dev`

closed 2016-09-05 by dlech