ev3dev/ev3dev #609: Downgrade kernel

cho934, 2016-04-17:
Hi,

Is it possible to downgrade to the previous kernel ? for exemple from 
https://github.com/ev3dev/ev3-kernel/releases/tag/v3.16.7-ckt26-10-ev3dev-ev3_1
to 
https://github.com/ev3dev/ev3-kernel/releases/tag/v3.16.7-ckt21-9-ev3dev-ev3_1
?

What are the correct instruction to downgrade ?

BR,

dlech, 2016-04-17:
if `flash-kernel` only ran once since you upgraded, the there should be `uImage.bak` in `/boot/flash` (the FAT partition). Replace the current `uImage` with this and it should boot the old kernel.

If `flash-kernel` was run more than once, the backup will be the same a `uImage`, so this won't work. You should be able to run `flash-kernel 3.16.7-ckt21-9-ev3dev-ev3` to reinstall that version.

cho934, 2016-04-17:
In fact, when I tried to flash ckt21-9, ckt26-10 is installed instead...
```
robot@ev3dev:~$ uname -a
Linux ev3dev 3.16.7-ckt26-10-ev3dev-ev3 #1 PREEMPT Sun Apr 10 23:32:19 CDT 2016 armv5tejl GNU/Linux
robot@ev3dev:~$ sudo flash-kernel 3.16.7-ckt21-9-ev3dev-ev3
[sudo] password for robot:
flash-kernel: installing version 3.16.7-ckt26-10-ev3dev-ev3
Generating kernel u-boot image... done.
Will use /dev/mmcblk0p1 as boot device.
Taking backup of uImage.
Installing new uImage.
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
robot@ev3dev:~$
```

dlech, 2016-04-17:
Maybe you have to use the `--force` option.

    flash-kernel --force 3.16.7-ckt21-9-ev3dev-ev3

cho934, 2016-04-18:
Perfect ;)
```
robot@ev3dev:~$ sudo flash-kernel --force 3.16.7-ckt21-9-ev3dev-ev3
flash-kernel: forcing install of 3.16.7-ckt21-9-ev3dev-ev3 instead of 3.16.7-ckt26-10-ev3dev-ev3.
flash-kernel: WARNING: Installing any new kernel package might override this.
flash-kernel: installing version 3.16.7-ckt21-9-ev3dev-ev3
Generating kernel u-boot image... done.
Will use /dev/mmcblk0p1 as boot device.
Taking backup of uImage.
Installing new uImage.
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
```

Then I reboot, and I can use my previous program using ckt21-9 kernel version.
I think this can be documented in the wiki.

closed 2016-04-18 by cho934