ev3dev/ev3dev #610: mailing list exists?

kortschak, 2016-04-18:
I have just written a [more complete Go interface](https://godoc.org/github.com/kortschak/ev3/ev3dev) to the ev3dev API from the descriptions given in http://www.ev3dev.org/docs/drivers/. Some of the information there appears to be out of date with respect to attribute paths (I'm using ev3dev-jessie-2015-12-30) and I don't want to spam here to get the updated information. So is there a mailing list for communication or another source of reasonably compact information around attribute paths and their meanings?

e.g. (but possibly not limited to)

- `speed_regulation` does not exist in http://www.ev3dev.org/docs/drivers/tacho-motor-class/ but is visible in tacho-motor file systems of my EV3 device.
- `stop_action*` in tacho-motor appears to have been renamed to `stop_command*`.

dlech, 2016-04-18:
We don't have a mailing list, so this is it. (I'm also usually on #ev3dev on irc.freenode.net).

If you missed the announcement, we just [released a new kernel](http://www.ev3dev.org/news/2016/04/11/Kernel-Release-Cycle-10/). This is why what you are seeing in the docs does not match what you see on the EV3. The documentation is not out of date, your kernel is. You need to update your kernel (`sudo apt-get install linux-image-ev3`) and update your Go interface to match the changes in the tacho motor class.

It looks like you already have found <http://www.ev3dev.org/docs/drivers/tacho-motor-class/>, which is about as compact as it gets.

kortschak, 2016-04-18:
Great. Thanks for the rapid reply. Should I file an issue to add my package to the extra languages section of http://www.ev3dev.org/docs/libraries/? I can either ping this when fixes are in or file an new issue.

dlech, 2016-04-18:
Any of those are fine. Or you can make a pull request to <https://github.com/ev3dev/ev3dev.github.io>.

Also, I would suggest adding something unique to the name of your project to differentiate if from the other go libraries.

* <https://github.com/ldmberman/GoEV3>
* <https://github.com/barnex/ev3dev-lang-go>

kortschak, 2016-04-18:
I'll send a PR. There is no name conflict between those project and mine though; the ev3dev API package is called ev3dev and imports via github.com/kortschak. Go namespacing is pretty well established in the community.

thanks

closed 2016-04-18 by kortschak

kortschak, 2016-04-18:
>The documentation is not out of date, your kernel is. You need to update your kernel (`sudo apt-get install linux-image-ev3`) and update your Go interface to match the changes in the tacho motor class.

linux-image-ev3 does not appear to exist in the repo - installed by `sudo apt-get install linux-image-3.16.7-ckt26-10-ev3dev-ev3` instead.

dlech, 2016-04-18:
Oops, it's actually `linux-image-ev3dev-ev3`. It's a meta package that will just install the same kernel package that you already installed.