ev3dev/ev3dev #611: Is there interest in speeding up start time of Node.js applications on ev3dev?

WasabiFan, 2016-04-19:
Recently, I have seen multiple [reports](WasabiFan/ev3dev-lang-js#7) of slow Node.js start-up times on the EV3. After I was contacted last week by @mkpk with some questions about this, we had a discussion about potential ways to help remidiate this problem. We came up with a few potential options and I wanted to open this up to the ev3dev community to see if anyone has any input. Mainly, I'm wondering if there are people that are using Node.js and are affected by performance issues: before I start anything, I'd like to have a sense of how people would benefit and what they would like to see.

As a first step, I think I will do some research into the specific tasks that Node is performing on initialization. I'd like to understand what it is that takes so long (as opposed to something like python, which is much faster) and see if there is something I could disable with a custom compilation to fix it. 

Past that, one of the more interesting possibilities that we had considered was creating some sort of a simple "Node service": one would start it in the background, and it would run a Node process and keep it warmed up for use when needed. Instead of using the `node` binary to run your script, you'd use a custom app that would just pass the target script to be evaluated by the already-running process in the proper context. That way, you'd only have to wait for the start-up time once, and it wouldn't even be blocking. Although I have a technical concept for one way of doing this, implementing it would require investigation into several potential solutions, including an external process that hosts Node and a JavaScript script that runs inside it. If I do decide to implement this, I would try to make it as generic as possible so it wasn't localized to ev3dev.

But before I do this, I'd like to get input from people who might use this to see what they think. I want to make sure that there are enough people using Node on ev3dev to justify the (hopefully minimal) investment, and make sure there are no technical limitations to consider. I know from the number of downloads of my ev3dev Node module that people _do_ use Node, but how many exactly is harder to gauge. In the interest of reaching this target audience, I'm opening this issue here so that I can make sure that the larger ev3dev community sees it.

mkpk, 2016-04-19:
Thank you for posting the idea. So I'm of course interested in its realization, if you find it makes sense. Anyway, regardless of your decision I intend to use node.js on my ev3 for my projects. And many thanks to all of you for your efforts related to development of this amazing project!

johngpayne, 2016-07-11:
Out of interest, what is the startup time for node like on an EV3? I've been working on something using "twisted" in python as a web service and for some reason starting up a very simple twisted server takes a very long time too, but its not clear why...

WasabiFan, 2016-07-11:
Node takes anywhere from four to eight seconds, although it's usually closer to four or five seconds.

johngpayne, 2016-07-11:
Thanks - that's a lot quicker than twisted / python (few minutes before the server appears). Will look to switch over.

dlech, 2017-10-03:
Is there anything to be done here? It seems like node is kind of a dead end on EV3 (debian dropped node on armel in stretch so ev3dev-stretch doesn't ship with node).

WasabiFan, 2017-10-03:
Agreed, there's no recovery for Node.js given that all support has been dropped.

closed 2017-10-03 by WasabiFan