ev3dev/ev3dev #612: Problems in the process to upgrade to Kernel 10

jabrena, 2016-04-19:
### System information:

* I am using: LEGO MINDSTORMS EV3
* I installed ev3dev using this image file: *ev3-ev3dev-jessie-2015-12-30.img*
* My kernel version is: *3.16.7-ckt21-9-ev3dev-ev3*
* My host computer is running: *OS X*
* I am connecting using: *Wi-Fi*

### About my issue:

#### What led up to the situation?

I tried to upgrade the kernel to last version and I executed apg-get dist-upgrade

#### What exactly did you do (or not do) that was effective (or ineffective)?

The system detected a new upgrade for kernel and I decided to upgrade the kernel but in the process I received the following messages:

``` bash
Setting up linux-image-3.16.7-ckt26-10-ev3dev-ev3 (1) ...
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory

 Hmm. There is a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build

 Hmm. The package shipped with a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source

Running depmod.
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 3.16.7-ckt26-10-ev3dev-ev3 /boot/vmlinuz-3.16.7-ckt26-10-ev3dev-ev3
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 3.16.7-ckt26-10-ev3dev-ev3 /boot/vmlinuz-3.16.7-ckt26-10-ev3dev-ev3
update-initramfs: Generating /boot/initrd.img-3.16.7-ckt26-10-ev3dev-ev3
flash-kernel: deferring update (trigger activated)
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 3.16.7-ckt26-10-ev3dev-ev3 /boot/vmlinuz-3.16.7-ckt26-10-ev3dev-ev3
run-parts: executing /etc/kernel/postinst.d/zz-flash-kernel 3.16.7-ckt26-10-ev3dev-ev3 /boot/vmlinuz-3.16.7-ckt26-10-ev3dev-ev3
flash-kernel: deferring update (trigger activated)
Setting up avahi-daemon (0.6.32-1ev3dev1) ...
Installing new version of config file /etc/avahi/avahi-daemon.conf ...
Setting up flash-kernel (3.35ev3dev8) ...
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_GB.utf8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_GB.utf8)
Setting up rtl8188eu-modules-3.16.7-ckt26-10-ev3dev-ev3 (5+1) ...
Setting up linux-image-ev3dev-ev3 (25) ...
Processing triggers for libc-bin (2.19-18+deb8u4) ...
root@ev3dev:/etc/apt# uname -r 
3.16.7-ckt21-9-ev3dev-ev3
````

I shutdown the brick but the previous version is on the brick. I repeat the process again in this case, the system recognize that it doesn't need to upgrade anything:

``` bash
root@ev3dev:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... The following packages were automatically installed and are no longer required:
  linux-image-3.16.7-ckt19-8-ev3dev-ev3 rtl8188eu-modules-3.16.7-ckt19-8-ev3dev-ev3
Use 'apt-get autoremove' to remove them.
Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ev3dev:~# sudo apt-get install linux-image-ev3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-image-ev3
```

Later, I checked packages:

``` bash
root@ev3dev:~# dpkg --get-selections | grep ev3dev
ev3dev-archive-keyring				install
ev3dev-connman-config				install
ev3dev-rules					install
ev3devkit-data					install
libev3devkit-0.1-0				deinstall
libev3devkit-0.3-0				install
linux-image-3.16.7-ckt16-7-ev3dev-ev3		install
linux-image-3.16.7-ckt19-8-ev3dev-ev3		install
linux-image-3.16.7-ckt21-9-ev3dev-ev3		install
linux-image-3.16.7-ckt26-10-ev3dev-ev3		install
linux-image-ev3dev-ev3				install
rtl8188eu-modules-3.16.7-ckt16-7-ev3dev-ev3	install
rtl8188eu-modules-3.16.7-ckt19-8-ev3dev-ev3	install
rtl8188eu-modules-3.16.7-ckt21-9-ev3dev-ev3	install
rtl8188eu-modules-3.16.7-ckt26-10-ev3dev-ev3	install
root@ev3dev:~# uname -r
3.16.7-ckt21-9-ev3dev-ev3
``` 

How to enable the last version?
Why I received this messages:

``` bash
Hmm. There is a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build

 Hmm. The package shipped with a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source
```

#### What was the outcome of this action?

#### What outcome did you expect instead?

I would like to execute uname-r and receive: 3.16.7-ckt26-10-ev3dev-ev3 

Many thanks in advance

jabrena, 2016-04-19:
I found a solution:

``` bash
root@ev3dev:~# sudo flash-kernel 3.16.7-ckt26-10-ev3dev-ev3
flash-kernel: installing version 3.16.7-ckt26-10-ev3dev-ev3
Generating kernel u-boot image... done.
Will use /dev/mmcblk0p1 as boot device.
Taking backup of uImage.
Installing new uImage.
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
root@ev3dev:~# shutdown -r now
root@ev3dev:~# uname -a
Linux ev3dev 3.16.7-ckt26-10-ev3dev-ev3 #1 PREEMPT Sun Apr 10 23:32:19 CDT 2016 armv5tejl GNU/Linux
root@ev3dev:~# uname -r
3.16.7-ckt26-10-ev3dev-ev3
```

Take a look the comments:

``` bash
Hmm. There is a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build
 However, I can not read it: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build

 Hmm. The package shipped with a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source
 However, I can not read the target: No such file or directory
 Therefore, I am deleting /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/source
```

jabrena, 2016-04-19:
How to know if latest modules are running:

``` bash
rtl8188eu-modules-3.16.7-ckt26-10-ev3dev-ev3
```

dlech, 2016-04-19:
>    locale: Cannot set LC_CTYPE to default locale: No such file or directory 
>    locale: Cannot set LC_MESSAGES to default locale: No such file or directory
>    locale: Cannot set LC_ALL to default locale: No such file or directory

This looks like you tried to upgrade the locales package and it failed. Try `sudo dpkg-reconfigure locales`

> Hmm. There is a symbolic link /lib/modules/3.16.7-ckt26-10-ev3dev-ev3/build
>      However, I can not read it: No such file or directory
>      Therefore, I am deleting [...]

This is normal and happens on every kernel upgrade. We don't ship kernel source code and I have never taken the time to figure out how to modify the packages to suppress this message.

> I found a solution

`flash-kernel` is supposed to run automatically when you install a new kernel, so if it does not, we have a bug.

> How to know if latest modules are running

`uname -r`. If the kernel version is `3.16.7-ckt26-10-ev3dev-ev3`, then so are the modules. Linux refuses use modules that don't match the kernel version.

dlech, 2016-04-19:
Can anyone else confirm that flash-kernel `3.35ev3dev8` is not running automatically when installing a new kernel?

Or perhaps is is because you did a `apt-get dist-upgrade` without `apt-get upgrade` first, causing `flash-kernel` and the kernel package to both be upgraded at the same time.

jabrena, 2016-04-20:
Hi @dlech, I executed apt-get dist-upgrade first.

dlech, 2016-04-20:
Yes, I could tell from the terminal output you posted. :wink:

Generally it is recommended to run...

    sudo apt-get update
    sudo apt-get upgrade
    sudo apt-get dist-upgrade

That goes for any Debian/Ubuntu system, not just ev3dev.

jabrena, 2016-04-20:
Many thanks for the help. :)

closed 2016-04-20 by jabrena