ev3dev/ev3dev #613: what does /dev/input/event1 (13, 65) reflect?

kortschak, 2016-04-20:
Just started playing with buttons and was wondering what event1 reflects events in. Is there documentation for this?

dlech, 2016-04-20:
Not exactly sure where you got that from. But I can say that the in ev3dev, the buttons on the EV3 are mapped to standard keyboard keys (Up, Down, Left, Right, Enter, Backspace) and use the standard Linux `evdev` (event device) interface. Just search for "evdev" and you can find documentation.

Also, I noticed that there are at least a couple of evdev Go libraries already that you could probably take advantage of.

kortschak, 2016-04-20:
I found /dev/input/event1 (major and minor device numbers 13 and 64) when I was investigating button handling and I was just wondering what physical device it corresponds to. My guess the usb port (which I guess I could have checked).

Yes, I saw the other libs. This project is partly fun and learning and partly due to the scope of those libs - they hide more than I want hidden.

dlech, 2016-04-20:
I like to use the `evtest` command to check out what the event devices (from the `evtest` package if it is not already installed). You can also find more info about them in `/sys/class/input`

dlech, 2016-04-20:
all of the numbers (`1` in `event1`, device major/minor, etc.) are meaningless. The important part is the device name. For example:

    $ cat /sys/class/input/input0/name 
    EV3 buttons

This is how you know you have the EV3 buttons and not something plugged into the USB port. `/sys/class/input/input0/capabilities/*` tells you what the device can do. In this case, it has key events and 6 different keys (as bit flags). And `/sys/class/input/input0/event0/dev` should be the same as `/dev/event0`.

kortschak, 2016-04-20:
It maps to "EV3 speaker beep".

Is there a doc page for the "input" class? I don't see anything in http://www.ev3dev.org/docs/drivers/.

dlech, 2016-04-20:
No, there are no docs for this on the ev3dev site because this is a standard Linux subsystem and there is plenty of documentation elsewhere.

kortschak, 2016-04-20:
Cool. Thanks for your help.

closed 2016-04-20 by kortschak