ev3dev/ev3dev #614: Sensors not recognized on BrickPi+

pambrose, 2016-04-23:
### System information:

* I am using: BrickPi+ V2.7

* I am using this image file: ev3dev-jessie-2015-12-30

* My kernel version is: 4.1.21-10-ev3dev-rpi #1 Sun Apr 10 21:50:56 CDT 2016

* My host computer is running: OSX

### About my issue:
When I plug in a sensor into my EV3, I see a new sensorX dir created in /sys/class/lego-sensor.
However, when I plug a sensor into my BrickPi, no sensorX dir is created.

I tried multiple sensors on multiple sensor ports and I rebooted the RPi.

I looked at the output of dmesg and no entries result from plugging/unplugging a sensor.

Motors are working better than sensors. When I plug/unplug a motor I still do not see a 
corresponding entry in dmesg, but /sys/class/tacho-motor/motorX dir properly reflects the motor status.

dlech, 2016-04-23:
motors and sensors cannot be automatically detected on BrickPi.

https://github.com/ev3dev/ev3dev/issues/529#issuecomment-177677082

closed 2016-04-23 by dlech

dlech, 2016-04-23:
I updated the sensors page to explain this better

http://www.ev3dev.org/docs/sensors/#using-sensors

pambrose, 2016-04-23:
Very helpful. Thanks.